ramapcsx2/gbs-control #179: Mis-matching output resolutions

ThreadOfFate, 2021-01-27:
I use a OSSC as my VGA to HDMI adapter.
When I'm running 50Hz games I have noticed that the resolution I set in the web UI does not match the resolution the OSSC is saying it's getting.
I have seen the OSSC say it's receiving 626p instead of the expected 576p.
Anyway to fix this? These odd resolutions basically force the OSSC to run in passthrough mode.

I'm seeing waves noise in the image, which I can fix by manually changing the number of lines in the web UI developer tab, it's only happening with the odd mis-matching output resolutions, but is probably unrelated noise.

lovrenco, 2021-01-29:
Hmm, are you posting issue not related with gbs-control firmware? gbs-control runs on modified Gonbes GBS-8200 boards, not OSSC (Open Source Scan Converter)

ThreadOfFate, 2021-01-29:
I'm aware, the GBS outputs via VGA natively, so since I already had an OSSC in my setup so, I'm using it to convert to to a digital signal.
The issue I'm having is the OSSC, tells me the resolution it is recieving is larger than the one I am setting the GBS to output.
This is only happening on 50hz games via the GBS and 60hz games are display the correct resolution, as do 50hz games plugged directly into the OSSC.
I'm saying this to make it clear, that I do not think the issue is from the OSSC.

ramapcsx2, 2021-01-29:
PAL has 625 active lines, which many "240p" consoles either bump to 626 or lower to 624 to get rid of the interlacing.
Your OSSC is reporting the correct number of lines.
What you're thinking of the the "visible lines", which are 576 for PAL.

ThreadOfFate, 2021-01-31:
I'm only using the GBS for my PS2, so that still seems odd.
So it's a 576i being deinterlaced targeting a 576p output and the OSSC is reading 626 lines.
I'm guessing it it had 1 less line the OSSC would just cull the non-visble lines.
This is only happening with 50hz games, withs 60hz works perfectly.