ramapcsx2/gbs-control #180: Htotal not saving with presets

AndehX, 2021-01-27:
So I think it's established that the Htotal settings in the Development tab directly affect the interference patterns than you get (especially at 1080p)  The standard 1080p preset has noticable interference that is fixed by adjusting Htotal+/- by a few increments.  The problem is that these adjustments are not saved to custom presets.

Can this be added?

amerika13, 2021-04-10:
I do not have issues with 1080p and interference, however, I did with 480p. Adjusting the Htotal up by 1 fixed that issue for me.

It would be nice if these did save with Presets as it seems like people might need to adjust their GBS-C setup individually with values that will not be consistent for everyone.

ferolim, 2021-04-13:
Hi,

In my case, what I did was to modify the code to apply the desired HTotal difference automatically.

I did the following (don't know if it is the best approach, but works):

Search for handleWiFi(0); inside the loop() function and add the following code right below:

//custom code
  rto->autoBestHtotalEnabled = false;

Then, add the following code below line 413 (after delay(1);)

//custom code
  if(rto->inputIsYpBpR == 0) {

    rto->forceRetime = 1;
    SerialM.println(F("Ajustando HTotal para RGB"));

    uint16_t VPERIOD_IF = GBS::VPERIOD_IF::read();
    static uint16_t VPERIOD_IF_OLD = VPERIOD_IF; // for glitch filter
    int valueToSubtract = 3;

    if ((VPERIOD_IF_OLD != VPERIOD_IF) || VPERIOD_IF == 522 || VPERIOD_IF == 524 || VPERIOD_IF == 526 || 
    VPERIOD_IF == 622 || VPERIOD_IF == 624 || VPERIOD_IF == 626) 
    { 
       valueToSubtract = 5;
    }

    SerialM.print(F("Utilizando o seguinte valor para o ajuste: "));
    SerialM.println(valueToSubtract);

    applyBestHTotal(GBS::VDS_HSYNC_RST::read() - valueToSubtract);
  }
  /////////////////////

Adjust the valueToSubtract accordingly. This code detects if the input is 240p or 480i on RGB and apply differente values for each cenario.

Hope it helps!

ramapcsx2, 2021-04-13:
Note that such a change has implications to all the timing routines. Best make sure interlace/progressive switches still work, and note that it won't have an effect on clockgen installed versions.

ferolim, 2021-04-13:
Hi!

I use my GBS with a PS1, N64, PS2 and Wii and had no problems. I also have the clockgen installed.

Do you have any tips on a better way to apply the HTotal and store its value?

Thanks!

amerika13, 2021-04-13:
My friend and I created a somewhat similar work around in a fork since dev settings were saving to presets with oversampling either. My setup looks very green with default OSR and it looked good on 2. So we changed the code so that it would default to 2 and added a place to change the value easy in the future if I need.

https://github.com/ramapcsx2/gbs-control/compare/master...lauradissinger:master

ramapcsx2, 2021-04-13:
About the OSR ratio, the ideal setting depends a lot on the source format. It's fine to override for individual installs, but best not override it for everyone ;p

@ferolim 
I just remembered that the clockgen mod fixes the timing by changing the display out clock, not the htotal value.
Your change should be fine with that, so no need to change it :)

amerika13, 2021-04-13:
@ramapcsx2 Yeah, that is why we did a fork for myself and am not going to ask to commit the change. It will very much depend on setup.