ramapcsx2/gbs-control #181: Wrong colors On Amiga

Foul, 2021-01-28:
I made all the hack proposed but the colors are weird.. here is a screen : https://img.super-h.fr/image/q3V05

Real colors must be purple for background..  with my eyes I see brown color instead.

It seems blue is missing.. ( I checked 2 times wires on removed potentiometers..)

My Amiga is connected to the 8pin RGBS input.

lovrenco, 2021-01-29:
Not software issue. Check your wiring. 
I built unit several days ago and colors are OK on A500 and A1200.
Bigger issue is that DblPAL resolution from the Amiga does not work correctly, and you cannot have 800x600 or 1024x768 output resolution in software.

ramapcsx2, 2021-01-29:
More output presets increase development workload 10 fold. I don't have any time as it is :p 

As for the colors, that is clearly a swapped color line, as @lovrenco says.
Swap blue and green, I think.

Foul, 2021-01-29:
after some research.. something wrong with blue RGB.. when i link Ground and red :working ground/green working... ground/blue :nothing .. nothing is moving with or wihout stripe on the Blue potentiometers (sorry for my language)

lovrenco, 2021-01-29:
@Foul I am using just one ground (video ground from RGB23). Basically I reused Commodore video cable, 9pin.
This is connector on monitor side:
https://imgur.com/yoUSfYJ
So only pin 1 ground is used, R,G,B lines and pin3 CSync (with 680 ohm resistor). 
I built cable for GBS board with female DB9 connector, attached it and it works.

Foul, 2021-01-29:
The Cable is not a problem, i use the same (own made) cable since many years and all was working before moding the GBS.

mine : https://img.super-h.fr/images/279f0eb4c88c2bdd8bf7b64c64f21706.jpg

Foul, 2021-01-30:
Does anyone know where the blue line from the 8 pin connect is going? what pin does this line go to? I need to bridge this line.

Foul, 2021-01-31:
a good weld and it's settled!

closed 2021-01-31 by Foul