ramapcsx2/gbs-control #183: Wifi Connectivity is lost if incompatible preset/setting is applied

lovrenco, 2021-01-29:
I know it sounds strange, but that is exactly what happens if I apply some frequency that gives "sync out of range" on monitor. 

Device disconnects form local network, I need to reset it few times to be able to access not through IP, but its default network

lovrenco, 2021-01-30:
I tested today. I used "Source pass through" option, monitor went out of sync and ESP (Wemos D1) disappeared form LAN (no ping), and his default network appeared, so I had to connect to it and reset options to get Wifi back.

andriiburka, 2021-02-23:
My device is disconnecting like 2 mins right after turning ESP on...

claesbert, 2021-05-31:
This happened to me too, but only if the wemos is directly on top of the GBS8200, it works much better if it is next to it! (probably the interference is just too much for the little wemos to handle!) Maybe there is a way to hack an ethernet port onto it for better connectivity?

ramapcsx2, 2021-05-31:
Ethernet is hard.
What you want to try is an ESP8266 module with external antenna connector + an antenna that's mounted a bit away from the scaler.

claesbert, 2021-05-31:
I just started researching that and it seems the wemos D1 mini pro has that option! Will try that one for sure!

xperiments, 2021-05-31:
Also check you are providing at least 2amps, happened to me when I was powering it with lower amperage