ramapcsx2/gbs-control #186: too much red on output- how to fix, or manually bias R G B input or output value(s)?

iainjh, 2021-02-01:
Hi

1st, thank you for such a great project!

Now my question. For some reason, my GBS8200 outputs what seems to be an excess of Red.

Using a GBS8200 v4.0, the 1280 x 1024 preset, and GBS control applied as per the github guides.

Atari ST Pal, input over RGB scart, has a pink cast over its white border and white areas on screen

On Amstrad 464, PAL RGB, the red appears severly overdriven. 

If I check auto gain is off, and increase gain manually, I can get rid of 50-75% of the pink for the ST.

But. at any level of gain I observe red tinging, and when scanlines are on (although these say for 240p) the scanlines show evenly for green and blue areas but NOT for red areas that might even show blurred  (see example pics x2, showing the red bias and the lack of scanlines in Red).  Is my GBS overreading the red in the RGB input?

Can I change any value in the sketch or .h files to bias the input or output values, to see if I can bring this excess of Red, down? I'm happy to experiment.

Thank you!  Regards, Iain

[image]

[image]

Arnold101knight, 2021-02-03:
how you made to work pal stuff? not works for me. you used some settings in particular?

iainjh, 2021-02-03:
i just selected a pal preset ie pal_1280x1024.h

btw I have noticed, if in the UI you click to force 60hz, there appears to be no way to switch it back to 50. in that case, i just erased flash in Arduino tools and made sure i didnt click that option. just select the 1280 x 1024.

In the log window, check it says its a 50hz preset.

can anyone help re the original post please.

0C-34, 2021-02-21:
> Now my question. For some reason, my GBS8200 outputs what seems to be an excess of Red.
>
> But. at any level of gain I observe red tinging, and when scanlines are on (although these say for 240p) the scanlines show evenly for green and blue areas but NOT for red areas that might even show blurred (see example pics x2, showing the red bias and the lack of scanlines in Red). Is my GBS overreading the red in the RGB input?

Hi! Some basic questions just to understand:
1) did you remove the color potentiomenters?
2) what display are you using? are you sure it isn't the display's fault? have you tried swapping with any other monitor?

iainjh, 2021-02-21:
---
> Hi! Some basic questions just to understand:
> 
> 1. did you remove the color potentiomenters?
> 2. what display are you using? are you sure it isn't the display's fault? have you tried swapping with any other monitor?
--

Hi! thanks for replying. 

My responses:
1. Yes, I removed the pots. I since checked the holes and traces, and remade the jumpers.  All seems OK.
2.  Its a Dell 2001FP, and I also tried it using a modern AOC 1440p monitor. Same issue on both.

As a temp workaround, I have since I dug around the gbs-control.ino file and hacked the red channel values to bias them down. I have also reduced the red levels on the dell to compensate somewhat. However this hasnt changed the no-visible scanline effect on red. from the images above, what are your thoughts?

Is there a UI setting planned to allow the R G B values to be altered?

Perhaps it is my board; I read they can vary in quality(?)

regards! 

Iain

xybots1967, 2023-11-16:
Hi I have the same problem I’ve got 2 gbs control systems I have built one is perfect and the other is displaying pink screen with image does anyone know how to get round this thanks 

[image]