ramapcsx2/gbs-control #188: Documentation suggestion: pinouts, tested modes and syncs, etc

claunia, 2021-02-02:
Hi,

I've been trying to connect an Amstrad CPC to a GBS with the gbscontrol firmware and I'm not sure if I'm doing things properly or not.

So, I've searched for documentation and found none.

I think it would be good to add the following pieces to the wiki for documentation:

1. VGA input pinouts. The IBM VGA standard has defined a pinout, enhanced by SVGA (ID pins) and later VESA (DDC/I2C pins). However some monitors support not so standard pinouts, mostly in reference to SYNC signals. There's no information at all if the GBS supports these variants or not, and many non-PC computers used them (e.g. Apple II, Amstrad, etc): CSYNC on HSYNC pin, Sync-On-Green.

2. TTL input. How to input TTL signals, including but not limited to, RGBIHV (CGA) and RRGGBBHV (EGA).

3. Supported sync frequencies range. What's the maximum and minimum, both horizontal and vertical, supported by the board, and the firmware? Changes with any mod?

4. Supported resolution modes. VxH@VR-HR. I tried with a very standard 640x480@60x31k and failed. Didn't have a way to know if I soldered the mod wrong, the GBS board was wrong, or anything, had to dig thru github issues.

5. Known cables. When you look for RGB cables schematics for old computers and consoles you see quite a few passive components and sometimes even active components (like sync splitters). Can I just pass the RGB signals from the input or do I need to add the components a SCART/PERITEL RGB-compatible TV expects (even AV input pin, or 4:3/16:9 selection pin)? Is it the same passing the signals thru the VGA input than thru the 5 metal pins on the board or the small white connector next to the VGA input?

gingerbeardman, 2021-02-02:
The wiki should be editable. Is it not for you?

claunia, 2021-02-02:
> The wiki should be editable. Is it not for you?

I don't have the answers for any of those questions :woman_shrugging:

gingerbeardman, 2021-02-02:
Can you edit the wiki or not?

claunia, 2021-02-02:
> Can you edit the wiki or not?

Yes, I can

gingerbeardman, 2021-02-03:
Then feel free to add your changes!

issalig, 2021-04-07:
Hi ,@claunia, I could not visualize the RGBS of a CPC with gbs-control but I have a configuration (288p) that is working not perfect but ok even without the SYNC-GND resistor.
You can find my code at https://github.com/issalig/gbs2cpc  Just flash it to an ESP8266 or arduino and set P8 jumper.

It would be nice to add this setup to gbs-control and it would be probably quite easy.

ramapcsx2, 2021-04-07:
Hey,
the problem with here is that gbscontrol is actively monitoring and controlling all the important chip parameters.
If you take a static configuration, then the sync monitor in gbscontrol will immediately overwrite the values.

To fix the sync problems in gbscontrol, the auto logic has to be extended (or fixed), so that it does the right things again and the sync stays stable.

Now, I realize that I created this issue in the first place. The code is sprawling and there are few consistent concepts.
Hence, it will be difficult for anyone else to go in there and work on the problem.
However, people that are dedicated enough will surely be able to do so. It's not *that* bad :p 

If you can parse this code and just need some tips to get started, please feel free to ask. I'll try to explain then.

issalig, 2021-04-07:
Thanks for the info. I do not know anything about gbs-control but flashing it :)  so I guess it will be harder than expected, 
I just found a code working on a Launchpad and converted it to ESP8266.

I think this will be a task for a core developer but I will take a look at it.

claunia, 2021-04-10:
Hi,

I have started this table with the tests I could do:
https://github.com/ramapcsx2/gbs-control/wiki/Tested-video-modes

It's a massive number of modes, and a massive numbers of don't work unfortunately.