ramapcsx2/gbs-control #189: Modes for 5:4 aspect ratio displays

shift838, 2021-02-04:
I would like to see resolutions for the standard 5:4 aspect ratio monitors.  not all of us want a widescreen display and would like to use this on resolutions like 1024x768 for some older computers like the Geneve 9640

LackofTrack, 2021-02-05:
Isn't 1280x1024 a 5:4 resolution?

shift838, 2021-02-05:
Yes, but the monitor I have has a resolution of 1024 x 768 max.

t9999clint, 2021-02-10:
I have a few displays that can do 1600x1200
It'd be pretty wicked to have that as a option. It also allows for easy integer scaling for a bunch of different resolutions

shift838, 2021-02-10:
I agree that more than one 5:4 aspect ratio should be offered.  This would expand the options for gbscontrol even further for users.

zajc3w, 2021-05-19:
second that, just spend a day modifying 1280x1024 to 1024x768@60HZ
It's a bit of a bodge, most parameters didn't make sense to me but it works, at least with 50hz rgbhv source
Try moving screen in web GUI if it appears chopped up after loading.

[pal_1280x1024.zip](https://github.com/ramapcsx2/gbs-control/files/6507373/pal_1280x1024.zip)

ramapcsx2, 2021-05-19:
If you spend 2-3 days more, you have a good shot at this! :)

mpalpha, 2025-02-24:
https://github.com/mpalpha/gbs-control