ramapcsx2/gbs-control #192: Irem M72 compatibility

bodgit, 2021-02-15:
I've successfully used a GBS-C with a number of arcade boards, (Capcom CPS-I/II/III, MVS MV1B, etc.), but I've tried to use it today with an Irem M72 board and the GBS-C refuses to find a stable sync signal. It correctly detects the RGBS input and I briefly see the LED on the ESP board flash (it normally stays on when a signal is detected) but then it stays off. Based on other bug reports here I enabled the "Print Infos" option and captured some of the log. I'm not sure how to interpret the output but I'm hoping it makes sense to you:
```
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:40 D:058c m:0 ht:2157 vt: 809 hpw: 167 u:318 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:40 D:0500 m:0 ht:2155 vt: 809 hpw: 167 u:319 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:40 D:058c m:0 ht:2155 vt: 272 hpw: 166 u:31a s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:40 D:058c m:0 ht:2154 vt: 270 hpw: 167 u:31b s: 0 S:13 W:-65
h: 430 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:00 D:058c m:0 ht:2156 vt: 541 hpw: 167 u:31c s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:00 D:058c m:0 ht:2156 vt: 541 hpw: 167 u:31d s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:00 D:058c m:0 ht:2157 vt: 273 hpw: 167 u:31e s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.30 H-   I:00 D:058c m:0 ht:2156 vt: 536 hpw: 167 u:31f s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2157 vt: 267 hpw: 167 u:320 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:0584 m:0 ht:2160 vt: 267 hpw: 167 u:321 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2154 vt: 268 hpw: 167 u:322 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2157 vt: 269 hpw: 167 u:323 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:0584 m:0 ht:2157 vt: 541 hpw: 167 u:324 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:0508 m:0 ht:2152 vt: 537 hpw: 166 u:325 s: 0 S:13 W:-65
h: 430 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2157 vt: 269 hpw: 148 u:326 s: 0 S:13 W:-65
h: 430 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2154 vt: 269 hpw: 167 u:327 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2153 vt: 269 hpw: 167 u:328 s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2154 vt: 269 hpw: 167 u:329 s: 0 S:13 W:-65
*h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2154 vt: 806 hpw: 166 u:32a s: 0 S:13 W:-65
h: 430 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2158 vt: 806 hpw: 166 u:32b s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2152 vt: 812 hpw: 167 u:32c s: 0 S:13 W:-65
h: 431 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2156 vt: 269 hpw: 167 u:32d s: 0 S:13 W:-65
h: 430 v: 567 PLL:0 A:a2a2a2 S:07.10.20 H-   I:00 D:058c m:0 ht:2154 vt: 269 hpw: 166 u:32e s: 0 S:13 W:-64
```
I have managed once by fluke to get some sort of picture however it was very unstable with a lot of tearing and then I think the code watching the sync line kicked in, tweaked the settings slightly, and I lost the picture again.

Any ideas?

ramapcsx2, 2021-02-15:
I can only tell that it struggles to separate VSync. There's no good fix that I can point to at this time, but if you search through older issues, maybe you can find something.

bodgit, 2021-02-16:
I did some enquiring on an arcade forum and apparently the M72 outputs 256 vertical pixels as opposed to the normal 240/224. Would that be the reason it's not locking onto the sync signal? If so, can the software be updated to handle that or does the hardware fundamentally not support such a signal?

ramapcsx2, 2021-02-17:
It's more about the format of the CSync (combined HS + VS) that some arcade boards produce. There can be some quirks to it that are just hard to deal with.
A software update might be able to fix this board, but it might break some others..

bodgit, 2021-02-17:
Would it help to capture the sync signal on an oscilloscope?

I presume you don't care too much about all of the hsync pulses in the middle of the screen, it's the vsync pulse at the bottom/top of the screen that is the troublesome area?

bodgit, 2021-02-17:
I think I managed to capture it:
[image]
From my limited understanding of csync I was expecting to see inverted hsync pulses during the vsync so I hooked up my CPS2 which works fine:
[image]
That doesn't have them either, but I notice the vsync pulse is about half as wide as the M72, (the BX-AX value). Finally I tried my MVS which also works:
[image]
 That _does_ have the hsync pulses, and the vsync pulse is about the same duration as the CPS2. Does that help at all?