ramapcsx2/gbs-control #194: VGA 640x480p60 source downscaled to 15khz

datamattsson, 2021-02-19:
I just got one of these boards built and trying to downscale a VGA source 640x480p60 to 15khz in an arcade cabinet. This is what I end up with (video):

- https://www.dropbox.com/s/mfsxjr30ulgk7wr/PXL_20210219_054200629.mp4?dl=0

The board builder told me that downscale only works with 480i which I confirmed by inserting a GBS-8100 480p to CGA 480i conversion PCB into the chain. The image is very washed out (white) even with the lowest of gains in the gbscontrol board. I rather not use the GBS-8100.

Any ideas?

radabass, 2021-03-15:
I'm having the same issue and I think is the same as in https://github.com/ramapcsx2/gbs-control/issues/128

I will try to add the XNOR sync combiner but I thint this may be the solution

datamattsson, 2021-03-15:
Thanks! This looks promising. Any "over-the-counter" solutions for RGBHV to RGBS? All I find is schematics.

radabass, 2021-03-16:
Hi @datamattsson,

It is confirmed, I've tried yesterday on a livestream to implement the sync combiner circuit and it works flawlessly.

The source was a 480p Dreamcast providing RGBHV and the gbs-control was downscaling to 15kHz 240p

As you can see from the live event screenshot without the sync combiner I have the same problem as you:

[image]

And after implementing the XOR Sync Combiner [https://www.epanorama.net/circuits/syncbox.gif](https://www.epanorama.net/circuits/syncbox.gif)

[image]

Everything works as expected, I am using an OSSC because I don't have a crt on my minilab but the ossc clearly states it is a 15kHz signal progressive outputting from gbs.

[image]

Danexmurder, 2021-03-16:
Would love to see an image of how you wired it up. It's usually easier to tell than by just looking at a schematic.

Danexmurder, 2021-03-16:
Seems like this would potentially do the trick using the parts mentioned in the schematic. 

https://oshpark.com/profiles/Gcharlie84

radabass, 2021-03-18:
Here you have the wiring, I put this board inside a DB15 female connector and used the P11 header.

[image]

3.2V/3.3V sync with these values.

datamattsson, 2021-03-18:
Someone build me one of these! :-)

gingerbeardman, 2021-03-18:
@radabass  I'm getting the components together for the diagram you have posted above, but having trouble finding the exact measurement capacitors. can I go more or less? i'm a bit of a noob. i have done a bunch of build but always of kits ready supplied, as well as a bunch of modding and fixing. thanks!

radabass, 2021-03-22:
I've compared the 240p outputting from the Dreamcast via scart and the gbscontrol 480p to 240p and it seems like the black lines between scanlines are a lot thickier than expected, also some text is not readable, I can see in the menu options for scanlines (I suppose by scanlines they reffer to the black lines between real scanlines) I there any way to disable the black "scanlines" completely in 240p?

radabass, 2021-03-25:
I have some pictures of the temporary setup, this works as intended.

The only problem I see are the big fat black lines between scanlines, **is it possible to change them somehow?**

[image]
[image]
[image]

Danexmurder, 2021-04-08:
I wanted to report back that I didn't have any luck using this board.

https://oshpark.com/profiles/Gcharlie84
There was no BOM listed so I set it up using all the parts from this.

https://www.epanorama.net/faq/vga2rgb/vga2rgbs_ttl.gif

Any advice would be much appreciated.

necroware, 2021-04-08:
@Danexmurder this adapter seems to deliver power from DSUB pin 9. Most of the devices don't use this pin at all and also don't deliver/consume power on that pin.

radabass, 2021-04-08:
I think @necroware is right, @Danexmurder you can try cutting the trace of pin 9 and soldering the 5v from the GBS to the chip VCC (or 3.3v, check your IC manual), I think some attenuation for the output of the IC may be good using a resistor on the 13th pin VGA output, I lowered the output to around 3.3v just in case but maybe it is not required for this board.

mmartinezespichan, 2021-05-27:
Hello, I have compile some configuracion for XBOX 360 and Dreamcast using the some information on web, I share with you please give me feedback
[gbscontrol.zip](https://github.com/ramapcsx2/gbs-control/files/6550539/gbscontrol.zip)

gingerbeardman, 2021-05-28:
@mmartinezespichan I have no idea what your files are trying to show. can you please explain?

mmartinezespichan, 2021-06-01:
Yes, I'm building a case for the GBSControl, I'll use for xbox 360 and dreamcast, for that I need to downscaled to 15hz and rescale to 240p. 

In order to set the goal, I have read that I need to first have another input for RGBHV in order to support it in the GBS-8200.

Also, I need and output for my arcade Sega Astro City, I did the following schema, but I don't know If It is only the way to get it or if I have an issue with the schema that I need to consider.

[image]

Icelvlan88, 2024-01-11:
Was a PCB ever put together for this?

KabukiFlux, 2024-01-26:
I'm sorry I have to remove my previous comments but I will leave here the PNG and the Fritzing file for documentation:

[image]
[sync_combiner.zip](https://github.com/ramapcsx2/gbs-control/files/14061642/sync_combiner.zip)

KabukiFlux, 2024-01-26:
And of course images on the test install on 2021:
[image]
[image]
[image]