ramapcsx2/gbs-control #197: esp8266-01s: "GBS-Board not responding"-> Remapping i2c-Ports possible?

uchristo, 2021-03-01:
Hi there,

I compiled the INO for Use with esp8266-01s and the WebFrontend and Debug-Pin seem to work fine. But the i2c-connection does not seem to work. When selecting a resolution in the GUI, the response is always "GBS Board not responding".
I wired the Debug-Pin to RX (GPIO3) and SDA to GPIO0, SCL to GPIO2 ( ~/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/variants/generic/pins_arduino.h )

`#define PIN_WIRE_SDA (0)
#define PIN_WIRE_SCL (2)

static const uint8_t SDA = PIN_WIRE_SDA;
static const uint8_t SCL = PIN_WIRE_SCL;

static const uint8_t RX   = 3;`

I measured continuity from ESP8266 GPIO0 / GPIO2 to Mason R10/R37 and even swapped the pins in the pins_arduino.h
I swapped the i2c-frequency from 400 to 700kHz:
`void startWire() {
......
  //Wire.setClock(400000);
  Wire.setClock(700000);
}` 

These are the settings within Arduino IDE:
[image]

As far as I understood the documentation, the default-PINs for i2c are GPIO2 for SDA and GPIO14 for SCL. ESP8266-01S doesn't offer easy access to GPIO14. Before soldering a wire to GPIO14, I'd like to know if swapping i2c-Port to GPIO0 and GPIO2 is supported.
https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf

uchristo, 2021-03-01:
Code-Markup doesn't work.

ramapcsx2, 2021-03-01:
That's a good question, but I don't have the answer. It shouldn't be too hard to find out though :)

uchristo, 2021-03-01:
According to https://forum.arduino.cc/index.php?topic=427607.15 it should be possible. But my findings suppose the opposite.

void startWire() {
  //Wire.begin();
  Wire.begin(SDA, SCL);
  // The i2c wire library sets pullup resistors on by default. 
  // Disable these to detect/work with GBS onboard pullups
  pinMode(SCL, OUTPUT_OPEN_DRAIN);
  pinMode(SDA, OUTPUT_OPEN_DRAIN);
  // no issues even at 700k, requires ESP8266 160Mhz CPU clock, else (80Mhz) uses 400k in library
  // no problem with Si5351 at 700k either
  Wire.setClock(400000);
  //Wire.setClock(700000);
}

This won't work for me. Either with 
~/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/variants/generic/pins_arduino.h:
#define PIN_WIRE_SDA (0)
#define PIN_WIRE_SCL (2)

nor with

~/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/variants/generic/pins_arduino.h:
#define PIN_WIRE_SDA (2)
#define PIN_WIRE_SCL (0)

gingerbeardman, 2021-03-01:
Use ``` above and below code to enable markup

uchristo, 2021-03-22:
Now bought a Espressif ESP-Wroom-02d with 4MB Flash. According to the Specs ( https://www.espressif.com/sites/default/files/documentation/esp-wroom-02u_esp-wroom-02d_datasheet_en.pdf , Chapter 3.4  Interface description) one should use GPIO2+GPIO14 for I2C. tried that by changing the definitions for SDA and SCL in /variants/generic/pins_arduino.h 
Still "GBS Board not responding"

Changing SDA and SCL to GPIO4+GPIO5 did the trick......
there are 2 possibilities:
a) The "General purpose" is correct just for the Pins without a special function
b) At some point, the defines SDA and SCL are not used?

However, initial problem seems to be solved

closed 2021-03-22 by uchristo