ramapcsx2/gbs-control #198: Atari-ST low res screen randomly jumps

xybots1967, 2021-03-04:
using the gbs-8200, using a GBS-Control, with my Atari-ST with Low Res screen the picture randomly jumps, im using a sync stripper using a LM1881, just wonder if any one has any tips to try and make the picture more stable.

the quality is awesome, ive tried different monitors. but the still randomly jumps 

Any help would be good

thanks

https://user-images.githubusercontent.com/80036989/109950178-2922f580-7cd4-11eb-92b0-c4f654c4bf16.mov

gingerbeardman, 2021-03-04:
At which timestamps in the video is the problem? Sorry I cannot see what you mean.

xybots1967, 2021-03-04:
Hi it jumps at 0:09 - 0:10 secs, I have other videos it jumps a lot what resolution should I use I have the clock generator installed as well

thanks

xybots1967, 2021-03-04:
hi

I've done another video timestamp 11,12,24,30 you see the picture jump, on my Super Snes the picture is perfect. And I've tried another RGB lead still the same, any help would be grateful.

https://user-images.githubusercontent.com/80036989/110042318-8737f300-7d3d-11eb-97c3-b14d348e571d.mov

gingerbeardman, 2021-03-05:
I see it now, sorry I am not familiar with this game to know what is expected or not.

xybots1967, 2021-03-05:
Hi 

its not the game it does it even on the desktop exactly the same jumping

ramapcsx2, 2021-03-05:
Hey,
you can try looking at your sync circuit, maybe browse some Atari forums for ideas.

xybots1967, 2021-03-06:
Hi

last night I tried a Atari Mega ST computer no modulator and the picture was rock steady, does any one know of a good guide for the STFM I could follow. thanks for your help and at least the GBS_8200 and GBS control are working as they should be.

claunia, 2021-04-12:
@xybots1967 can you share the schematics of the cable lead, if you know it?

And fill the missing data on the table I've put on https://github.com/ramapcsx2/gbs-control/wiki/Tested-video-modes ?

Thanks

NullArgument, 2023-07-08:
I was experiencing some drop outs in my video where I could tell that the esp was rebooting by the LED. After I put 150R resistors inline with the RGB signals just before going to the GBS it went away.

Do you have any resistors on the RGB lines going into your GBS?