ramapcsx2/gbs-control #201: VGA output stays active for some time after powering down an input

goutsune, 2021-03-09:
So basically after powering down my PS2 or Wii in 15khz mode (it doesn't affect EDTV) my CRT stays on for about a minute or two. With RGBS it displays just black (sometimes with few lines of original picture) and with component/BW composite I get green screen, this happens on any default preset as well as my custom ones.

P.S. I'm having this issue since quite an old revision, and since nobody reported it I have a feeling something is wrong with by setup.

ramapcsx2, 2021-03-09:
Is that a problem?

goutsune, 2021-03-09:
Well.. it just feels weird as there is going to be no further activity, the room is black at night but monitor stays on with green screen (I usually use component to have progressive supports). :)
If it is essential for further signal detection to work, I have no objections except for perhaps somehow blanking the screen.

ramapcsx2, 2021-03-09:
I don't remember all the motivations I had for setting the timeout, so I'd not want to change it based on one request.
The code is there however, and you're free to set any timeout you want for your build.

https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L5886
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L5942
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L6639
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L6754

Change all 4 hardcoded "0x07fe" to something lower, maybe 0x05fe.

This is pretty bad code, and I'm sorry for this. I wanted to reuse this variable a lot :p