ramapcsx2/gbs-control #207: Question: Is it possible to adjust brightness and contrast and colors from the web UI?

lovrenco, 2021-03-19:
When I look picture from Amiga 1200 on VGA 17' monitor using GBS scaler with gbs-control, colors are "washed" comparing to same machine outputting to PAL monitor, or PAL TV using SCART. 

I tried tuning that on monitor itself with some success but I wonder is there a way to tune image  brightness and contrast and colors form the scaler itself?

Thank you.

gingerbeardman, 2021-03-19:
What is the status of the RGB pots on your GBS?

Have you tried Gain+/- or Auto Gain Adjust?

lovrenco, 2021-03-20:
Gain+/- does the trick. But more control would be better, if something like that is even possible. 

> What is the status of the RGB pots on your GBS?
> 
> Have you tried Gain+/- or Auto Gain Adjust?

gingerbeardman, 2021-03-20:
Did you adjust or remove the RGB potentiometers on your GBS PCB?

lovrenco, 2021-04-25:
> Did you adjust or remove the RGB potentiometers on your GBS PCB?

I removed them.

closed 2021-04-25 by lovrenco

gingerbeardman, 2021-04-25:
Was that your solution?