ramapcsx2/gbs-control #209: Midway Arcade PCB Compatibility

Happymealtm, 2021-03-25:
Using a GBS-8200 with the GBS control, debug pin attached, all pots removed and hard wired to the left. Standard Arcade PCBs work fine, Midway games such as Mortal Kombat 2 etc will not sync. It detects the RGB input but then keeps looking and swaps to no input.

I have no idea how to relate the stock standard GBS menu to this new one. To fix this on a stock GBS-8200 you would go to Geometry Menu and then adjust the V position to the value 28 and that's pretty much all you changed. But I have no idea what the Geometry Menu and V position of 28 related to in this new firmware.

Is there a way to manually change this? Letting it auto search doesn't pick it up. It is out of sync and can barely see the picture.

Thank you

ramapcsx2, 2021-03-25:
These games are using very far off spec vertical timings, right? Something like 56Hz or so?
If the auto code doesn't work on them, there isn't much you can do.

fluxcorenz, 2021-03-25:
I bought my GBS8220 from jammaboards.com back in 2013, which has a modified firmware to support the midway boards. Indeed, it runs my UMK3 board fine. Unfortunately it sounds like the owner of jammaboards passed away, and the business also.

Last time I tried, GBSC couldn't find the signal, but when I removed the debug pin (disabling GBSC), the original firmware took over and UMK3 worked again.

This implies the GBS is capable of working with the Midway boards, if controlled correctly...

ramapcsx2, 2021-03-26:
That's sad news, yeah :/

The hardware itself is flexible enough to support 56Hz sources fine. The problem is doing all the gbscontrol tasks on it reliably.