ramapcsx2/gbs-control #213: Composite & Svideo - Question, not a feature request

amerika13, 2021-04-11:
Hello!

I got curious today and decided to hook up a console via Svideo (PAL cd32) to my GBS-C that is using component input just to see what it would actually do. To my surprise, it works fine. It obviously won't display in color and is B&W due to how component works vs. how Svideo works. But it recognized the sync signal and I get a clean stable image.

Could the GBS-C technically work with composite and svideo input if it was programmed to do so? Or is there some hardware/technical hurdle that limits it's input options.

Thanks!

Nuck-TH, 2021-04-12:
To work with composite, you need PAL/NTSC decoder, TIVA chip can't do that.
S-Video is closer to component(luma is same) than composite, but i guess it will need additional circuity to split colors from chroma.

ramapcsx2, 2021-04-12:
Yeah, there's no hardware on the GBS for this. An external decoder to analog RGB would be required, and that adds additional processing that might make the end result too bad overall.

amerika13, 2021-04-12:
Ok, that makes sense. I assumed it was one of two things. Either there wasn't a care/big enough use case to enable composite/svideo or the hardware simply didn't support it. Looks like it was the latter.

Thank you for the responses and closing!

closed 2021-04-12 by amerika13