ramapcsx2/gbs-control #214: Scrambled Image from PS1, PS2 and PS3

SephirothFan, 2021-04-12:
Hello!  I'm not sure if this is the right place to post this particular issue, but hopefully someone can help me out.

Here's my setup and what I'm trying to do.

I have my PS1 connected through RGB JP-21 Sync on Luma to an Extron Crosspoint (which is all CSYNC).  That then goes out over RGBS to my PVM 1342Q and XRGB-mini Framemeister.
I have my PS2 and PS3 connected through Component into the same Extron Crosspoint.  That then goes out to a RetroTINK COMP2RGB transcoder, which then goes to my PVM.  The Component output also goes from the Extron Crosspoint to the Framemeister's Component input.

What I would like to do is patch RGB from my PS1 and/or Component from my PS2/PS3 to the GBS-8200 and then back out to my Extron so that I can send the final signal to my PVM and Framemeister so that the Framemeister doesn't drop the 240p/480i switch when I stream Chrono Cross.

So, here's the issue.  I can hook everything up, but I get a scrambled image on my PVM and the Framemeister keeps getting a signal and dropping it.  I've tried hooking up everything the way that I understand I should which is:

RGB
PS1 RGB -> Extron -> Monoprice VGA to RGBHV cable (using just the gray line to get what I'm assuming is RGBS) -> GBS-8200 VGA Input -> Monoprice VGA to RGBHV cable back into the Extron (again just using the Gray or H line for RGBS) -> PVM and Framemeister

Component
PS2/PS3 Component -> Extron -> BlueRigger Component Cable -> GBS-8200 Component Input -> Generic VGA to Component Cable -> Extron -> Framemeister and RetroTINK Transcoder to RGB which goes to the PVM.

But, I'm not sure where the problem is.  I've tried using the 15KHz Downscaling option and that doesn't work.  I've also tried the passthrough option, but that doesn't seem to work at all and seems to freeze the GBS-8200 to the point where I have to unplug it and plug it back in to unfreeze it.  I also can't seem to get the transcoding options to work if I mix and match RGB and Component inputs and outputs.  Any suggestions on how to get this setup working?

Thanks!

ramapcsx2, 2021-04-12:
Eh, I don't have all these devices :)
What you can do is, try and figure out where the chain fails.
A PS2 via Component, a PSX via RGB, both definitely work well with gbsconstrol.
There's going to be a conversion somewhere that it doesn't like.

SephirothFan, 2021-04-13:
Thanks for the response!  I will try to simplify my setup again, check as many facets as I can, and I'll report back when I have something.  I have a feeling the Extron is the problem in the chain, since they can be kinda picky regarding the type of signal they can switch.

SephirothFan, 2021-04-15:
I have tested my equipment again and I can get a signal on my PVM now, but the picture is rolling, so I'm starting to think that it's a sync issue.  Does GBS Control still support RGBS over the D-Sub ports, because I'm only seeing RGBHV?

ramapcsx2, 2021-04-15:
CSync can be fed to the H pin, as usual.

SephirothFan, 2021-04-15:
Okay, I'm using a Sync on Luma cable from my PS1 which goes into a JP-21 to BNC Cable with a LM1881N Sync Stripping circuit inside; which converts Sync on Luma into CSync.  I'm then outputting that CSync signal into the VGA port of the GBS-8200.