ramapcsx2/gbs-control #215: How to revert to default chip to try 1024*768 resolution?

lovrenco, 2021-04-13:
Hello, 
Is there a way to revert to default GBS8200 firmware without desoldering ESP8266 and clock generator?
I would like to see if small 14 inch CRT will work with 1024x768 resolution. 
Check photos. In 720x480 picture is crisp and stable, but there is no way to get it to show on full screen, not with monitor hardware, nor with ESP software controls. 1280x1024 is too much for this monitor (as is for any 15 inch display).

I tried removing jumper, no luck. Is there a way I can try default firmware without removing all components?

[image]

[image]

ramapcsx2, 2021-04-13:
The jumper has to be opened, and the ESP SDA / SCL disconnected.

lovrenco, 2021-04-14:
I tried opening jumper, disconnecting ESP power line, but I get just a black screen, no video.

Nuck-TH, 2021-04-14:
You have to disconnect data lines too.

closed 2021-04-25 by lovrenco