ramapcsx2/gbs-control #216: GBS-C sync issue with Neo Geo

amerika13, 2021-04-20:
I recently integrated the GBS-C into my main capture chain and I am having a bit of an issue with my Neo Geo. It, for some reason, won't sync correctly.  You can see that easily in this video. I also added the debug terminal into the capture so you can see what is going on there.  https://www.youtube.com/watch?v=6pWeF0ECJFc

Neo Geo description: Neotropolis modded Neo Geo that adds, via jacks, composite, svideo, component and stereo to the back of the unit. It is a 3-6 and it is hooked up via component out.

I do have another Neo Geo 3-5 that is completely stock and fixed up recently that I can hook up via HD Retrovision cables to test, but that will probably be another day or so due to my time being pretty limited at the moment.

Thanks

ramapcsx2, 2021-04-21:
Hey, this is a known problem with the current sync watcher code when using certain MVS board revisions.
I need to debug it, but I don't have the time..

amerika13, 2021-04-21:
This is an AES unit if that matters. And I hear you on the time thing. Never enough of it to go around.

Sorry to re-post a known issue. I did search for open AES/MVS bugs but found nothing. Do you want me to close this or keep it open?

Thanks Rama and take care!

Ryoandr, 2021-05-05:
I tested several AES and MVS, and in my results some MVS are unable to sync and all AES were unable.
I sent Rama 2 MVS, one that worked and more recently one that didn't. Hopefuly he manages to look at it soon. :)
So be patient all ;)

AleffCorrea, 2021-05-08:
My Neo Geo CD also has issues with sync, I get "heatwaves" coming out of it (basically a jittery image), but at least with some patience turning the sync signal physically on and off again multiple times it seems like it does eventually sync "correctly" (you can see subtle scanline drift if you pay attention but it's minimal).
How do we bribe you to look into it? (:

jboffel, 2021-09-11:
Hello, I can confirmed the two AES I tried could not sync properly, a lot of skipping on the video and top of the screen always lagging. I used stock RGB with board 3-5 & 3-6.
For info the voltage for stock RGB of AES is offset by +2V.
I assembled the GBS-C following Voultar tutorial on YouTube.

jboffel, 2021-09-13:
I've a setup that allow me to input command to the serial while running the AES on a screen.
I'm not knowledgable enough to know what I'm messing with but if you get me what command you need to generate some valuable output to understand the issue, I can try to reproduce them.

jboffel, 2021-09-14:
Trying out more things, I realized the skipping is due to the syncwatcher activity probably trying to figure out the optimal setting. But for some reason it seems to fail to do so. When I run ! command in the serial I see the return value for sfr is 0 as well as pll.
The AES use a C sync at 15.75khz at stock. Can it be a reason here for not being able to get the pll?
Sorry I might say non sense as I'm just experimenting with my limited understanding of video signal.

ramapcsx2, 2021-09-14:
The problem is complex. You can read all about it here:
https://www.hdretrovision.com/blog/2018/10/22/engineering-csync-part-1-setting-the-stage
This machine is going to have one of the mentioned sync issues, and the sync processor / gbscontrol can't figure it out.

jboffel, 2021-09-26:
I've been running a few extra tests just to try eliminate possibilities, I've wired a LM1881N in between the AES 3-5, 3-6 stock to see what would happen if I feed a H sync and V sync. Based on my understanding it is not supposed to be necessary actually for the AES.
But anyway, I thought maybe there is a signal issue from what the AES is outputting at stock and what the GBS-control like.

I'm still forcing the  getPulseTicks() output to force a close by value of what the AES Jap is supposed to do (59.18Hz), this return value 2703616 gives: "source Hz: 59.18000 new out: 59.18000 clock: 108000000 (+0)"

I also need to toggle on Low Res: Use Upscaling, otherwise I don't have video output at all (also attached the log of the initial sync in that case)

But then with RGB/HV (CSync) and RGB/HV upscale mode I get the video signal reliably (mean each time I activate this setup it shows up). However I can still see this weird horizontal black line between each video line, as like it is skipping half of the horizontal lines. I can't find a setting that would avoid it. I attached the picture of the pattern from one corner of the screen.

[image]

Complete sync log with upscaling:
00:55:54.344 -> ext clockgen detected
00:55:54.379 -> <reset>
00:55:54.417 -> Chip ID: 1A 1
00:55:54.417 ->  G: 3E G: 3F G: 40
00:55:54.706 ->  R: 3E R: 3F R: 40 R: 41 R: 42
00:55:54.964 ->  B: 3E B: 3F B: 40 B: 41 B: 42
00:55:55.253 -> <reset>
00:55:55.286 -> Activity detected, input: RGB
00:55:55.502 -> VSync: present
00:55:55.502 -> HSync: present (with CSync)
00:55:56.140 -> RGB/HV (CSync) 
00:55:56.140 -> clock gen reset: 81000000
00:55:56.863 ->  RGB/HV upscale mode
00:55:56.863 -> 59.18
00:55:56.897 -> ADC offset: R:42 G:40 B:42
00:55:56.897 -> clock gen reset: 108000000
00:55:56.968 -> 
00:55:56.968 -> preset applied: 1920x1080 for scaling RGB (CSync)(set your TV aspect ratio to 16:9!)
00:55:57.005 -> Note: scaling RGB is still in development
00:55:57.005 -> 
00:55:57.324 -> source Hz: 59.18000 new out: 59.18000 clock: 108000000 (+0)

Without upscaling:
00:16:26.868 -> ext clockgen detected
00:16:26.941 -> <reset>
00:16:26.941 -> Chip ID: 1A 1
00:16:26.978 ->  G: 3E G: 3F G: 40
00:16:27.260 ->  R: 3E R: 3F R: 40 R: 41 R: 42
00:16:27.518 ->  B: 3E B: 3F B: 40 B: 41 B: 42
00:16:27.803 -> <reset>
00:16:27.840 -> Activity detected, input: RGB
00:16:28.060 -> VSync: present
00:16:28.060 -> HSync: present (with CSync)
00:16:28.688 -> RGB/HV (CSync) 
00:16:28.688 -> clock gen reset: 81000000
00:16:29.996 -> (SP) HS active low

ramapcsx2, 2021-09-26:
So what I take from this is that it helps to hardcode values for 59.18Hz resulting Vfreq?

jboffel, 2021-09-26:
Well to be honest I don't know if this is the reason. I just forced this value because it's supposed to be the expected input freq. But truly, it also works similarly with hard coded frequencies even up to 64Hz.
The main things being that if I don't force a returned value then vsyncInputSample() will return false then forcing getPulseTicks() to return 0, therefore failing completely to find a source Hz...

Ryoandr, 2021-09-26:
AES refresh rate is not the same as MVS. AES have a slightly higher crystal to be closer to NTSC (only the earliest AES are the same as MVS). 3-5, 3-6 have the higher crystal for sure.
https://wiki.neogeodev.org/index.php?title=Framerate

jboffel, 2021-09-26:
Thanks for the link, I'll try to force value that closely match the 59.599 to see if it makes any difference

jboffel, 2021-09-27:
Well, I made a fool of myself I'm afraid!
After diving more into the source code, I realized the debug pin used for the vsyncInputSample() seems to be exclusively use for that.
So I started to think what if my wiring is wrong and I actually did not soldered properly the PIN D6.
And guess what, of course it was that...
After soldering properly, I can say for sure that neo geo JAP NTSC 3-6 is working like a charm and is auto-detecting best setting: "source Hz: 59.60177 new out: 59.60181 clock: 161263664"
I was not so far with my manual setting previously from the truth, I'm ashamed to admit that the black lines was just the scanlines filter with low intensity, if I deactivate it the image is really nice.
However it is still my setup with LM1881N, I'm yet to test without the chip in RGB (Csync)...

jboffel, 2021-09-27:
Ok so I tried bypassing the LM1881N, and it is same result as before, unable to sync in RBG (Csync) without the V sync signal.
Is it only a software issue or a problem with the signal out of a stock AES I'm not sure, but it can't sync ATM without the LM1881N

Ryoandr, 2021-11-26:
So I tried an LM1881 on both the sync line and the csync out of the CXA1145 and I still couldn't get a stable picture. However, it made me look at AES video circuit, and It has a whole circuit that adjust system frequency to better set composite encoding.
https://wiki.neogeodev.org/index.php?title=Video_PLL

Jboffel, I'm assuming you did the 1881 sync stripping on the composite video line, so maybe it's the only solution.

That still wouldn't explain why some MVS also have sync problems...

jboffel, 2021-11-27:
If I remember correctly I just wired straight RGB cables to the upscaler then wired the composite video provided by the aes to the lm1881 video input then wired the vsync output to the vsync entry of the upscaler. Then use the upscaler with the RGB/HV instead of CSync... Of course I used usual resistor and capacitor with the lm1881 as well.

Ryoandr, 2021-11-27:
ah ok, you used 1881 Vsync output too, I didn't, gonna try

Ryoandr, 2021-11-27:
Ok so i tried lm1881 vsync, while still on cxa1145 csync output, even tried with a pot to see if a load changed something. Couldn't get a stable and reliable pic. So then I tried the 1881 on the composite video line...
It worked instantly...
Kinda annoying since I try to use clean signals to avoid crosstalk in cables... 
So now I have to think about an LM1881 implementation, in either the console or the gbs.

jboffel, 2021-11-27:
Personally I just added it in the scart cable gap back of the pins to avoid touching the upscaler or the console...

davince72, 2023-08-11:
> Ok so i tried lm1881 vsync, while still on cxa1145 csync output, even tried with a pot to see if a load changed something. Couldn't get a stable and reliable pic. So then I tried the 1881 on the [...]

Could you share schematics maybe or more detailed description. I recently bought the GBS-C and the new 161-in-1 multi cart.....just to find lot crap TV output ;-) In my case the screen output seems slowmotion with lots of scrambled output and the screen (if picture can be seen for a moment) it's shifted horizontally about 2/3. 
First I was shocked and thought my AES was broken ;-) Luckily my TV has scart input and that worked flawless.

Ryoandr, 2023-08-16:
@davince72 
```
                                        LM1881
                           HS---470---1         8-----+---+5V
Neogeo AV Out Pin 3 ---+---------||---2         7     |
                       |   VS---------3         6---+ | 
                      75R           +-4         5   | |
                       |            |               | |
                       +------------+-------+--680K-+ |
                       |                    +---||--+ |
                      GND                   +---||----+
			    
All caps 100nF			   
HS and VS can be either on the VGA or the JST 
```

Hendrix900, 2024-01-21:
Hello,
I had the sync problem on my Japanese AES 3-6.
I have managed to solve it without having to add lm1881.

The steps are:
=GBSC=
1-NEO GEO AES compatibility mode = off
2- Sync on Green pressed
=GBSC UI=
3-Activate developer mode
4-In the developer option, disable SyncWatcher

This will make the image run correctly.
When you disable SynWatcher, sometimes there may be some wrong horizontal lines at the top. playing to activate and deactivate the SyncWathcer end up disappearing.

I hope this can help
All the best

ramapcsx2, 2024-01-21:
Right, it's clear that the syncwatcher code is seeing something it thinks is wrong. If someone can figure it out, they can surely tweak that variable and fix it properly :)

Aimen06, 2024-07-09:
hello is there any update with this issue ? I have an mis 2 slot and the image randomly jump. :/ 
I have try to modify the  getPulseTicks() with 
`// Tentative d'échantillonnage de la vsync
    if (!vsyncInputSample(&inStart, &inStop)) {
        Serial.println("Utilisation des ticks de secours");
        
        // Utilisation de la valeur calculée correspondant à 59,1856 Hz
        return 101370; // Ce nombre est une valeur de secours basée sur la fréquence de 59,1856 Hz
    }`

But it doesn't work :/  
Is there something I can help to fix  it with neogeo mis /aes @ramapcsx2

Aimen06, 2024-07-09:
With no hope I have revert to  a previous commit  
https://github.com/ramapcsx2/gbs-control/tree/abecf525034e820b4ff1be91c41f246cb89ef16c 
And it works perfectly  directly with my mvs board mv2f .
Maybe be you can look what breaks   the compatibility with neogeo system

carlos-biker, 2024-07-10:
> With no hope I have revert to a previous commit https://github.com/ramapcsx2/gbs-control/tree/abecf525034e820b4ff1be91c41f246cb89ef16c And it works perfectly directly with my mvs board mv2f . Maybe be you can look what breaks the compatibility with neogeo system

Can you explain to me how I can install this version please? I have the same problem with a Neo Geo AES.

Aimen06, 2024-07-10:
Click on the link , download the repo .
then click on gbcontrol.ino (adruino ide much launch) .
Plug the esp on your computer and compile then upload  the firmware and enjoy.
I have tested 24h no more jump.

carlos-biker, 2024-07-11:
It doesn't work for me, when I turn it on, the light stays red and the small screen doesn't turn on.

Aimen06, 2024-07-11:
Did you made the gbs control or bought it already made ?

carlos-biker, 2024-07-11:
I bought it in AE.

Aimen06, 2024-07-11:
Contact the seller for upgrading it  , i don't which component it use. 
I have made it using the doc.

carlos-biker, 2024-07-12:
> Hello, I had the sync problem on my Japanese AES 3-6. (It has RGB Bypass with csync) I have managed to solve it without having to add lm1881.
> 
> The steps are: =GBSC= 1-NEO GEO AES compatibility mode = off 2- Sync on Green pressed =GBSC UI= 3-Activate developer mode 4-In the developer option, disable SyncWatcher
> 
> This will make the image run correctly. When you disable SynWatcher, sometimes there may be some wrong horizontal lines at the top. playing to activate and deactivate the SyncWathcer end up disappearing.
> 
> I hope this can help All the best

Hello, there is no way to save this configuration, is there? I mean, the only way is to disable SyncWatcher every time you turn on the console.

tiagoandriotti, 2025-04-11:
Same here with a NeoGeo AES revision 3-5 for a while, BUT in my case someone around managed to make it work and passed down the knowledge.

Some say in these units (3-5 and 3-6) c-sync and composite are mixed but, why? Don't ask me. I've been through hell to make it work here, I'm just passing ahead the data. Mine seems to be a stock Japanese unit, and I won't open it FGS.

I'm using GBS C like the one below and updated the firmware in April 2025, so, maybe something has changed and it saves all the configs I have left... even the other consoles work with it normally:
[image]

Cable - It is from a local Brazilian vendor. Seems like nothing special, but in the wiring the Scart pin 20 is linked to the composite pin, probably no resistor or signal cleaning circuit like in other cables because in this one the multimeter detects the connection (but don't ask me because I'm not a specialist). Just check in the multimeter if it detects the composite line going to pin 20 and it might be good to go.

[image]

On the GBS buttons, first with it powered off:
Compatibility mode (on the left) - ON
RGBS (on top, by the side of the HDMI) - ON
RGBS Sync (red button on the side) - ON
Sync on Green -  (red button on the side) OFF

Next, connect all the stuff and set the GBS on.
On the web interface, on the options tab, turn the Developer mode on.
Set it all the way you see in the picture below:
[image]

The FrameTime lock may be activated or deactivated, try it on or off.

Now for the most uncertain part of the game: the Developer tab.
I tried so many things before the fix that we didn't even manage to be sure how it was in the default because it was like that when we make it work so, it may look like this:

[image]
The instruction I had was to disable SyncWatcher but this button stuck in my interface many times and I'm not sure if it is on or off, but theoretically if yours look like this it might be good.

Every unit, cable and GBS have its own particularities, even the OSSC has its own tricks for making it happen, so, I hope this can help someone. Let me know if it works for someone else.

Chandler-Kluser, 2025-08-15:
I have just tried, now. The losing sync issue has ended, but now there is some isse with the very first lines:

[image]

[image]

[image]

The best solution so far is still [this one](https://github.com/ramapcsx2/gbs-control/issues/118#issuecomment-2636015597)

Chandler-Kluser, 2025-08-15:
@tiagoandriotti  this solution I pointed worked flawlessly without any glitch, the bad part is the older firmware, which does not support downscaling to 15kHz but I don't mind.

Be sure when you call "last GBS" to mention it is that bitfunx legacy blue model, there are tons of different models of GBS-C and my both models are not that one (mine are DIY).

Just a hint: if you are not sure if an UI button is pressed or not in your cell phone refresh the page and button states will be updated