ramapcsx2/gbs-control #217: Horrible Distortion trying stretch image with horizontal scaling

invaderlex, 2021-04-23:
I'm running some widescreen hacks in PS2 games and a PSP using component cables. When I try to use the function for horizontal scaling the image began distorting the more stretched the image is to the point of being a complete indistinguishable mess. Is this normal? The function was not meant to be used to stretch much? I tried installing a clock generator, but is the same. Happens in all resolutions.

gingerbeardman, 2021-04-23:
I also get this, I have come up with a system where I combine output of the GBS and do some extra scaling with an Extron DVS304. Not ideal but workable.

ramapcsx2, 2021-04-23:
That's normal, yea.
The scaling function allows users to correct the width a little, but it is constrained by what the chip can process.
Depending on the source format, it might not be possible to stretch it out even a little more.