ramapcsx2/gbs-control #220: 720p passthrough from Xbox not working correctly

dzastsed, 2021-04-29:
current setup: Original Xbox (motherboard 1.0; iND5003.06 bios) with component cables -> GBS 8220 -> VGA out to LG M1994D LCD TV
mods done to GBS: esp board (obviously), added external clock generator, ferrite bead and 1uf cap on R58 point, powered by 5V 1.5A PSU
issue: 720p content does not output in a stable form (keeps losing signal, some distortions every few seconds and such, also says that video out is at 61hz). Example here: https://streamable.com/h6n5z3 ; video out info here https://cdn.discordapp.com/attachments/606550060284510218/837364380269019216/IMG_20210429_192658.jpg
480i/480p stuff works excellently

ramapcsx2, 2021-04-29:
Might be the same problem as with your PSP.

Roy12321334, 2021-06-17:
https://github.com/ramapcsx2/gbs-control/issues/162

I had a similar issue. You should mod the XBOX with a TRUE VGA BiOS and then either use the native sync on green output or do the H/V hardware mod.

jerbearwright, 2021-12-17:
Update: Ive found that many of the vga to hdmi adapters have sync issues ive found one on amazon that solves this issue 

https://smile.amazon.com/gp/product/B00WAIX5S4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1