ramapcsx2/gbs-control #222: Downscaling 480p PS4 to 240p not working

danieldoyle, 2021-05-01:
For input: I have a PS4 set to output 480p, this is going into GBS via a Tendak HDMI => VGA adapter
For output: I have a vga to component passive cable

In the GBS control web UI, I can turn on dev mode and see it pick up the PS4 output activity:
```````````````<reset>
<reset>
Scanning inputs for sources ...
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV 
(H-PLL) rate: 4214 state: 5
RGB/HV 
(H-PLL) rate: 4214 state: 5
RGB/HV 
(H-PLL) rate: 4214 state: 5

But then I try to click on 15Khz and it just does that last entry and the TV is a mess like it can't sync.  I have component YPbPr enabled in settings.  A number of the UI elements in the web UI don't always seem to toggle either on the mobile UI or the desktop.  Not always, sometimes they work, but other times I can see the setting is applied in the dev output but the UI toggle element doesn't actually update.

gingerbeardman, 2021-05-02:
Are you using the new WebUI?

danieldoyle, 2021-05-02:
I just set this up last Friday so I believe it’s the latest.  The UI is very snazzy looking and not the big oval buttons I expected from many of the screenshots about the project

gingerbeardman, 2021-05-02:
Great, that gives @xperiments more to go on.