ramapcsx2/gbs-control #224: Arrow PSU's sold out

crabvacation, 2021-05-02:
[RetroRGB's video](https://www.youtube.com/watch?v=fmfR0XI5czI) on building and setting up a DIY GBS-C recommend that you purchase a 5v Power Supply from Arrow, [specifically this model](https://www.arrow.com/en/products/wsu050-4000/triad-magnetics), but it's out of stock on their website. Are there any other recommended PSU's? Sorry if this is a dumb question, the video also talks about how using the wrong PSU could potentially destroy the device, and nobody wants to break the thing they've been working on, y'know?

gingerbeardman, 2021-05-02:
Of course you can use any PSU, i think it's a bit of FUD from RetroRGB.

Previously I used an Apple iPhone with USB to barrel cable.
Currently I use a power extension USB socket with the same USB to barrel cable.

I find the USB barrel cable approach most versatile.

webhdx, 2021-05-02:
Don't be too hysterical over stuff :) Really cheap Chinese power supplies for sure have bigger chance of failing and possibly destroying your device. In reality you don't hear very often about failing power supplies. Just don't buy the absolutely cheapest stuff from AliExpress. If you are handy at soldering (I assume you are since you bought GBS) then get a beefy phone charger and replace the plug. Phone chargers coming with smartphones are usually high quality products.

crabvacation, 2021-05-06:
I appreciate the comments left, but I am mostly looking for a substitute I can just buy and use outright. I'm comfortable with soldering things like surface mount and throug-hole components on a PCB, but I am WAY to nervous and inexperienced to be comfortable making something that I'd be inserting into a wall outlet or USB outlet.

damopinn, 2021-06-08:
https://www.amazon.com/StarTech-com-USB2TYPEM-Barrel-Power-Cable/dp/B003MQO96U
this could be an option for you, if your usb adapter is upt to it.

NO-ob, 2021-09-03:
I'm using one of these with my phone charger it works well https://www.adafruit.com/product/4090

cathalferris, 2022-04-20:
Not all 5V power sources are the same.

One thing about the phone chargers, is that they can carry an *awful* amount of ripple current, at frequencies in the tens of kilohertz and repeated up through all of the shortwave range. I'm involved in amateur radio, and I have a few software defined radio devices. I have seen quite a few wall-to-USB adapters - even big name brands - absolutely spew radio energy into the surroundings.

Most devices on USB won't be very affected by this extra radio energy, but for sure some crosstalk will happen. The GBS may be one of those devices that will show moiré or other "walking" interference patterns from poor quality and poorly regulated and filtered power.

Just worth remembering when someone has specified a known-good known-working brand of PSU for a device.