ramapcsx2/gbs-control #225: 240p 120hz

Sirethief2525, 2021-05-03:
While this isn't an issue, more like a question or a feature request if anything. But would it be possible to display 240p at 120hz  instead of 60hz for those of us who use rgb home consoles on a vga monitor to get genuine scanlines? I know that it potentially comes with downsides but from what I've read and even tested with gameboy interface running in my gamecube through swiss, I can't see a difference or any of the issues others seem to have or be seeing. Thanks in advance for any information you can provide.

Sirethief2525, 2021-06-11:
From my understanding the old dooklink software had a 240p@120hz mode but that was powered by a raspberry pi instead of an arduino, but if someone is familiar with both types of coding the it could be possible to port over that preset to the arduino but I'm not the guy unfortunately

santiis2010, 2022-06-16:
Oh i hope this can be done, 240p with 120hz would be awesome, at least the creator of the firmware can answer for yes or no right?

Toxithot, 2022-06-19:
my primary use for the gbscontrol is to connect the PS1 to a crt monitor, and this would be so useful for the 90% of games that don't go into 480i mode. unfortunately i don't have any idea how to understand the resolution .h files, and the dooklink project uses a different equally incomprehensible format or i'd have already tried swapping stuff around to see what breaks.