ramapcsx2/gbs-control #226: Custom preset from known SYNC oscillogramme / timings

nightgryphon, 2021-05-05:
How can i build and upload custom preset in case i can record the exact syns sygnal with logic analizer and measure any part of this sygnal?
I have problems with GBS Control on some retro PC like "УКНЦ МС 0511" and some others. There is no or unstable picture. But i see the sync is valid and both TV and old tube monitor car recognize it. Can i manually create the required preset when GBS auto detect fail?

Here is the example SYNC graph captured with logic analyzer at the output of video processor before mixing
[image]

zajc3w, 2021-05-19:
I have similar problem, I have CNC screen to replace, 18.3khz h-sync,  50hz v-sync.
I get right half of the screen cut off.
TV5275 is poorly documented and changing output res to something useful for me(1024x768) took a day of hex editing.
If there is profile generator it's not known to duckduckgo nor google. all i got is register description PDF and programming manual, which isn't the greatest help.
now I'm stuck on input shifting.
I have some black empty on the left and right edge cut off.
No matter how I try to move it in web GUI I get green garbage on the right.
Signal works perfect on borrowed CRT and looks ok on oscilloscope - video signal is not overlapping h-sync polarities are correct.

ramapcsx2, 2021-05-19:
There is a serial command parser in the code. You send it characters and it does them.
You can use this to quickly set the main parameters of a preset, such as the number of lines, and it'll figure out what to configure to get them. Example: "wvt 1066" means "write vertical total 1066", which would produce enough lines for an x1024 preset.
Here is the parser, sadly undocumented (I wrote it for myself :p) https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L8230

ramapcsx2, 2021-05-19:
You also need to google how to read modelines, then use a modeline list (https://www.mythtv.org/wiki/Modeline_Database#VESA_ModePool) to find some main parameters of presets.
Your x768 preset would use 806 total lines, for example (wvt 806)