ramapcsx2/gbs-control #230: I cannot compile

akira62pate, 2021-05-17:
I have tried to compile GBS-C according to the Software installation Wiki with different Arduino Software Versions and different ESP8266 Boards because my version could not be recognized.

I got this error every time.

Arduino: 1.8.15 (Windows Store 1.8.49.0) (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\akira\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\akira\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\akira\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=10C4_EA60 -ide-version=10815 -build-path C:\Users\akira\AppData\Local\Temp\arduino_build_26089 -warnings=all -build-cache C:\Users\akira\AppData\Local\Temp\arduino_cache_742115 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.python3.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -verbose C:\Users\akira\Documents\Arduino\libraries\gbs-control\gbs-control.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\akira\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\akira\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.49.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\akira\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=160,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=10C4_EA60 -ide-version=10815 -build-path C:\Users\akira\AppData\Local\Temp\arduino_build_26089 -warnings=all -build-cache C:\Users\akira\AppData\Local\Temp\arduino_cache_742115 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.python3.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\akira\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -verbose C:\Users\akira\Documents\Arduino\libraries\gbs-control\gbs-control.ino

Using board 'd1_mini' from platform in folder: C:\Users\akira\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0

Using core 'esp8266' from platform in folder: C:\Users\akira\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0

Detecting libraries used...

"C:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/include" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lwip2/include" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\akira\\AppData\\Local\\Temp\\arduino_build_26089/core" -c -w -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10815 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\cores\\esp8266" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\variants\\d1_mini" "C:\\Users\\akira\\AppData\\Local\\Temp\\arduino_build_26089\\sketch\\gbs-control.ino.cpp" -o nul

Alternatives for Wire.h: [Wire@1.0]

ResolveLibrary(Wire.h)

  -> candidates: [Wire@1.0]

"C:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/include" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lwip2/include" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\akira\\AppData\\Local\\Temp\\arduino_build_26089/core" -c -w -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10815 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\cores\\esp8266" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\variants\\d1_mini" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\libraries\\Wire" "C:\\Users\\akira\\AppData\\Local\\Temp\\arduino_build_26089\\sketch\\gbs-control.ino.cpp" -o nul

Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]

ResolveLibrary(ESP8266WiFi.h)

  -> candidates: [ESP8266WiFi@1.0]

"C:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/include" "-IC:\\Users\\akira\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lwi [... cut ...]
      |                                   ^~~~~~~~~~~~~~~~~~~~~

C:\Users\akira\Documents\Arduino\libraries\gbs-control\framesync.h:36:22: note: declared here

   36 | void ICACHE_RAM_ATTR risingEdgeISR_prepare() {

      |                      ^~~~~~~~~~~~~~~~~~~~~

C:\Users\akira\Documents\Arduino\libraries\gbs-control\framesync.h:68:20: warning: 'void risingEdgeISR_prepare()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]

   68 |     attachInterrupt(DEBUG_IN_PIN, risingEdge

xperiments, 2021-05-17:
It seems you modified some way(by error, etc) the gbs-control.ino file and added "esp" at the begining of the document. Please remove it & try to recompile. Or better, re-download the project & recompile.

So where in your code line 1 says:

````esp#include "ntsc_240p.h"````

needs to be 

````#include "ntsc_240p.h"````

akira62pate, 2021-05-17:
Thank you for the Feedback but in the Line there is no esp# mentioned it is clear. I have tried it at second PC and had the same inssue. Could it be the 3.0.0 from the ESP driver link or the update from GitHub which can cause it?

#include "ntsc_240p.h"
#include "pal_240p.h"
#include "ntsc_720x480.h"
#include "pal_768x576.h"
#include "ntsc_1280x720.h"
#include "ntsc_1280x1024.h"
#include "ntsc_1920x1080.h"
#include "ntsc_downscale.h"
#include "pal_1280x720.h"
#include "pal_1280x1024.h"
#include "pal_1920x1080.h"
#include "pal_downscale.h"
#include "presetMdSection.h"
#include "presetDeinterlacerSection.h"
#include "presetHdBypassSection.h"
#include "ofw_RGBS.h"

akira62pate, 2021-05-17:
Problem solved, I used the 2.7.4 Version of the ESP8266 libary and iwth this no problem.

closed 2021-05-17 by akira62pate

gingerbeardman, 2021-05-18:
@akira62pate which version were you using before? I'd like to check it. tahnks

akira62pate, 2021-05-18:
3.0.0 - I normally try the latest one.

I still got these warnings 

C:\Users\akira\Documents\Arduino\libraries\gbs-control\gbs-control.ino:9608:9: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]

but I could upload it to the ESP and it runs.