ramapcsx2/gbs-control #231: Help needed for gbs-c downscaling

0C-34, 2021-05-18:
I have SCART CRT (RGBS 15KHz) and I want to connect an hdmi "thing" (in my case I want to connect a steam link, a nintendo switch and a raspberry pi4, all of which have 480p hdmi output capabilities).
I've seen a lot of people convert the input VGA (RGBHV ~31KHz) to RGBS usinc a Sync combiner, and then combine it again after downscaling it.
Since I've seen a lot of different circuit for it and I have NO WAY to troubleshoot them (I have a 25€ pocket oscilloscope at max :|) I wanted to ask what circuit(s) do you suggest for the sync combiner and how I should terminate and attenuate the video output.

ramapcsx2, 2021-05-18:
I'm sorry, I can't guide you through this.
But if you google VGA sync combiner, you should come up with some example circuits.
I think RetroRGB also has an example.

ramapcsx2, 2021-05-18:
Sync levels / amplitudes should be fine up to 3.3Vpp, which is the chip's IO voltage.

0C-34, 2021-05-18:
Really thanks a lot anyway for the response (and especially for the sync levels, I was sure all analog levels had to be 1Vpp.... for some reason) even if it's not the main scope of the gbs-c project.
One last thing before i close this issue: I know I may be asking the wrong person, but do you happen to know by chance where should I ask these questions?
I tried on the crt gaming reddit and the response i got was (and I kid you not) "You'll have to figure out how the GBS-C outputs and then match it up to the scart pinout"

ramapcsx2, 2021-05-18:
A lot of talk happens on Discord now. I'm not sure which channel best fits the question, but feel free to look around :)
https://discord.gg/JAaVemEQ

0C-34, 2021-05-18:
Thanks a lot again, have a good day :D

closed 2021-05-18 by 0C-34