ramapcsx2/gbs-control #232: can not compile sketch

undonevil, 2021-05-19:
i keep getting an error when trying to compile the latest software. here is the output. thank you

Arduino: 1.8.13 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\alex\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\alex\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\alex\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\alex\AppData\Local\Temp\arduino_build_20914 -warnings=none -build-cache C:\Users\alex\AppData\Local\Temp\arduino_cache_210719 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -verbose C:\Users\alex\Desktop\gbs-control\gbs-control.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\alex\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\alex\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\alex\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\alex\AppData\Local\Temp\arduino_build_20914 -warnings=none -build-cache C:\Users\alex\AppData\Local\Temp\arduino_cache_210719 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.mkspiffs-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.0-newlib4.0.0-gnu23-48f7b08.path=C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08 -verbose C:\Users\alex\Desktop\gbs-control\gbs-control.ino

Using board 'd1_mini' from platform in folder: C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0

Using core 'esp8266' from platform in folder: C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0

Detecting libraries used...

"C:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/include" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lwip2/include" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\alex\\AppData\\Local\\Temp\\arduino_build_20914/core" -c -w -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\cores\\esp8266" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\variants\\d1_mini" "C:\\Users\\alex\\AppData\\Local\\Temp\\arduino_build_20914\\sketch\\gbs-control.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for Wire.h: [Wire@1.0]

ResolveLibrary(Wire.h)

  -> candidates: [Wire@1.0]

"C:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/include" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lwip2/include" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\alex\\AppData\\Local\\Temp\\arduino_build_20914/core" -c -w -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\cores\\esp8266" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\variants\\d1_mini" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0\\libraries\\Wire" "C:\\Users\\alex\\AppData\\Local\\Temp\\arduino_build_20914\\sketch\\gbs-control.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]

ResolveLibrary(ESP8266WiFi.h)

  -> candidates: [ESP8266WiFi@1.0]

"C:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-newlib4.0.0-gnu23-48f7b08/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/include" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lwip2/include" "-IC:\\Users\\alex\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\alex\\AppData\\Local\\Temp\\arduino_build_20914/core" -c -w -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP82 [... cut ...]
Using library DNSServer at version 1.1.1 in folder: C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\DNSServer 

Using library ESP8266mDNS at version 1.2 in folder: C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ESP8266mDNS 

Using library ArduinoOTA at version 1.0 in folder: C:\Users\alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\ArduinoOTA 

exit status 1

call of overloaded 'abs(uint32_t)' is ambiguous

webhdx, 2021-05-19:
You are probably using ESP8266 core version 3.0.0 which is incompatible. Install 2.6.3 via board manager.

zajc3w, 2021-05-19:
call of overloaded 'abs(uint32_t)' is ambiguous
if (abs(inPeriod - outPeriod) <= 4)  in frame sync, both variables are unsigned, either cast them to signed or drop abs() depending if inPeriod ever gets lower than outPeriod.

saying that, downgrading to 2.7.4 fixes the compile error.

ramapcsx2, 2021-05-19:
That sounds like I added this to check for an overflow, yes.

undonevil, 2021-05-20:
Worked perfectly after downgrading to 2.7.4, thank you.

Also what is the relevant information so i can edit down the origonal post.

rabidgerry, 2021-05-20:
I hate to hijack this, but I had the exact same issue

call of overloaded 'abs(uint32_t)' is ambiguous
if (abs(inPeriod - outPeriod) <= 4) in frame sync, both variables are unsigned, either cast them to signed or drop abs() depending if inPeriod ever gets lower than outPeriod.

and now that I downgraded to ESP8266 core version 2.7.4

I get this instead, any ideas?

Arduino: 1.8.15 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\theme\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\theme\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\theme\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10815 -build-path C:\Users\theme\AppData\Local\Temp\arduino_build_508781 -warnings=none -build-cache C:\Users\theme\AppData\Local\Temp\arduino_cache_673767 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -verbose C:\Users\theme\Documents\gbs-control\gbs-control.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\theme\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\theme\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\theme\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10815 -build-path C:\Users\theme\AppData\Local\Temp\arduino_build_508781 -warnings=none -build-cache C:\Users\theme\AppData\Local\Temp\arduino_cache_673767 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -verbose C:\Users\theme\Documents\gbs-control\gbs-control.ino

Using board 'd1_mini' from platform in folder: C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4

Using core 'esp8266' from platform in folder: C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4

Detecting libraries used...

"C:\\Users\\theme\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\theme\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\theme\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\theme\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\theme\\AppData\\Local\\Temp\\arduino_build_508781/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 [... cut ...]
In file included from C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/Arduino.h:27:0,

                 from C:\Users\theme\AppData\Local\Temp\arduino_build_508781\sketch\gbs-control.ino.cpp:1:

C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/stdlib.h:16:20: fatal error: stddef.h: No such file or directory

 #include <stddef.h>

                    ^

compilation terminated.

exit status 1

Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

zajc3w, 2021-05-21:
Looks like you are using Arduino 15xx it's about 5 years old, try updating to 18xx.
If still fails make sure you have all dependencies installed as per wiki instructions

rabidgerry, 2021-05-21:
Hmmm not sure I follow you, says v1.8.15 which I actually updated to last night as I thought perhaps this was the issue as well.  I was using v1.8.10.  and had updated firmware lots of time before without issue.

rabidgerry, 2021-05-21:
There are two versions I seem to be able to download of the Arduino Software, one states for Windows 7 and above and the other is in App form for Windows 8-10.  I had both it turns out.  The Windows 7 and above version did not say Arduino IDE, just Arduino, it is this version I have been using regularly to update GBS Control.  

The other one does say Arduino IDE, I think I downloaded that for something else.  I've removed everything off my PC now and will try installing the v1.8.15 from scratch.

rabidgerry, 2021-05-21:
Ok it didn't work the first couple of times but I think I managed to get it working this time, does this look ok?

Executable segment sizes:
IROM   : 613192          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 29640   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1316  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 4928  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 27952 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 649076 bytes (62%) of program storage space. Maximum is 1044464 bytes.
Global variables use 34196 bytes (41%) of dynamic memory, leaving 47724 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:f4:32:2f:1a:06
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 653232 bytes to 506764...
Wrote 653232 bytes (506764 compressed) at 0x00000000 in 11.7 seconds (effective 448.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Stretchdude, 2021-05-24:
I changed  the call to abs() function in framesync.h:148 into labs() then it worked for me. Using 1.8.15 on Linux

undonevil, 2021-06-09:
thanks guys worked like a champ after switching versions

closed 2021-06-09 by undonevil