ramapcsx2/gbs-control #233: add serial command guide to wiki

zajc3w, 2021-05-19:
it's hard to guess syntax of sxxx and gxxx serial commands based on the code
@ramapcsx2 
can you please add human readable text list to wiki or codebase?
I'm trying to add/modify presets, and doing it on preset.h files and recompiling takes ages
Sxxx commands seem to be doing what I need by looking at he code.

ramapcsx2, 2021-05-19:
Sorry, I can hardly remember them, and this is a lot of effort to try and explain.
But here is the "set" commands. I hope I remember this right..
s5s07s42  sets segment 5 (check datasheet for the 5 segments) register 0x07 to value 0x42.
If you send this, the ADC gain for one color should change. Try a value of 0x7F

ramapcsx2, 2021-05-19:
That doesn't exist, not as far as I remember anyway. You'd have to code it.