ramapcsx2/gbs-control #234: Apple II Composite Input

chisaipete, 2021-05-21:
Fantastic work!

I'm considering diving into using my GB8220 v3 to solve this problem, but wanted some advice to find out if it's impossible or not 😁 

My source is the Apple //e composite/RCA video out signal. It has a [notoriously non-standard NTSC output mode](https://en.wikipedia.org/wiki/Apple_II_graphics). 

I have hooked up the RCA output to the GB8220 single Y-input (on the Y-Pb-Pr inputs). The result in YPbPr mode with the stock firmware is a slightly disappointing and fuzzy black-and-white monochrome output. I've hooked up the Apple //e to a new-ish Sony LCD Television and have seen similarly fuzzy fuzzy output, except with color! I know my Apple //e is certainly capable, it looks like some kind of synchronization issue (?) on the receiving end.

I'm looking at this project as a potential way to tune this synchronization through the firmware and get a cleaner, crisper, and color signal from the Apple //e to a modern VGA output device. Before embarking on this quest, is this theoretically possible with this modification and much effort, or is this a fool's errand?

ramapcsx2, 2021-05-21:
Hey,
it sounds like you want Composite Video decoding, which this hardware cannot do at all.
You need a scaler that can decode it, for example the Retrotink products.

chisaipete, 2021-05-21:
Ah bummer 😞 thanks for the idea, I'll check out your suggestion.

closed 2021-05-21 by ramapcsx2