ramapcsx2/gbs-control #237: Force PAL 50hz to 60hz doesn't do anything

Joshimuz, 2021-05-26:
Title says it all really, am I missing something?

I'm not the only one with this problem, after a quick search through the issues I found #155 which, while quite long winded, is basically the same problem I'm having.
The GBS-C just continues to output at 50hz with this setting on even post restart as confirmed by both my TV and my OSSC. Just nothing changes when I toggle this setting on. Is there some kind of detection for PAL sources that isn't working?
Not sure I can be bothered to post images and logs and stuff when #155 did the same thing and just got told "lol buy a display that works with 50hz".

If the setting doesn't work that's fine, it's free open source software I can't complain, but why have the setting? :D

gingerbeardman, 2021-05-28:
Which version of the GBS-C are you using?

YoImLee, 2021-06-01:
Do you have the clock generator board? I have it installed and the setting is ignored for me also. Not sure if it's intentional or not but I do think it's the culprit.

TimbleThimble, 2021-07-12:
Any luck with solving this? Trying to get 1080p PAL PS2 output (which apparently can only be done with the 50 - 60hz toggle on) but doesn't seem to do anything.

I also have clock gen.

kimmesantana, 2021-09-13:
I have the exact same issue. Without external clock generator everything works fine, as soon as I enable the external clock generator the ForcePal60Hz Option doesn't work anymore. 
I can't get any mode to output 60Hz.

rabidgerry, 2021-09-24:
I'm no expert but as far as I am aware when you install the clockgen mod, it simply outputs what it has coming in via input.  So it it's receiving a 50hz signal it can only output a 50hz signal and the same would be if receiving a 60hz it can only output what it is given.

fabbrimichele, 2022-05-29:
I have the same issue.

I am trying to output a PAL 50 Hz signal to VGA 60 Hz.
_Force PAL 50hz to 60hz_ works without the Si5351 installed but it's been ignored since I installed it.

(I like the old VGA monitor because it's 4:3 format)

fabbrimichele, 2022-05-29:
Perhaps it wouldn't be too complicated to have a setting to disable the external clock generator from the UI.

fabbrimichele, 2022-05-29:
I modified `externalClockGenDetectPresence()` to always set `rto->extClockGenDetected = 0` regardless of the presence of the external clock generator and force to 60 Hz works again. It would be great to make it configurable.

fabbrimichele, 2022-05-29:
I created a simple GIT patch that disable the External Clock Generator when Force PAL 50hz to 60hz is enabled:

```
 gbs-control.ino

 void externalClockGenDetectPresence() {
+  // When externalClockGen is present forcing to 60 Hz does not work.
+  // So if the setting is enable ignore externalClockGen.
+  if (uopt->PalForce60 == 1) {
+    return;
+  }
+  
   const uint8_t siAddress = 0x60; // default Si5351 address 

```

ramapcsx2, 2022-05-30:
Hey, that's all great, but maybe you can step it up and try to make it an option in the UI? Is that doable for you? :)

fabbrimichele, 2022-05-31:
@ramapcsx2 sure. I will try to make an option to disable the the external clock generator regardless of the PAL force to 60Hz option. It will take a bit because I am not familiar with the UI part.

ramapcsx2, 2022-05-31:
Of course.
If it's too hard to add an extra option, you can still add the clock gen disable code to an existing one.
Good luck! :)

fabbrimichele, 2022-05-31:
I'm Gonna try adding a new property :)

fabbrimichele, 2022-06-06:
Hi @ramapcsx2 I implemented the new option for disabling the external clock generator on both side: UI and back end.
I have local branch with the changes, should I create a git patch or should I create a pull request?

[image]

fabbrimichele, 2022-06-06:
@ramapcsx2 could you please give me the permission to push my branch?

ramapcsx2, 2022-06-06:
Ususally this goes via pull requests. You can do everything from the Github page. Does it work for you?

gingerbeardman, 2022-06-07:
@fabbrimichele you should do like this:

1. fork the project
2. put your changes on the fork
3. commit to your fork
4. create Pull Request

fabbrimichele, 2022-06-07:
Hi @ramapcsx2 and @gingerbeardman I created the pull request: https://github.com/ramapcsx2/gbs-control/pull/334