ramapcsx2/gbs-control #241: Confirming Correct Debug Pin Connection

shawneeunion, 2021-06-07:
I recently purchased a GBS 8220 with the GBS Control modifications. One thing that was missing was the debug pin connection to the ESP8266. I was able to quickly correct this but haven't seen much of a difference in performance or image quality. I'm using the GBS Control with a Nintendo Wii with component cables and a VGA to HDMI adapter. I'm wondering if there are any tests in the web GUI or other methods to confirm the debug pin is connected correctly. I sometimes need to barely shift the image to get it centered but it has not been a huge deal.

damopinn, 2021-06-08:
To test the clock generator is working, open the GBS Control web gui. Go to Preferences and scroll down to Activate FrameTime lock. Press on the FrameTime Lock button and the console will read “Active FrameTime Lock not necessary with external clock gen installed”. Video output should be pristine and free of horizontal tearing when high output resolution presets are used.
I copied that from the wiki.

Nuck-TH, 2021-06-10:
Debug pin and clock generator are two unrelated things.

damopinn, 2021-06-10:
ok, the debug pin is connected to pin 11 of the Trueview5725 chip so you could check for continuity to there from the esp8266 board,

shawneeunion, 2021-06-10:
Thanks, should of just tested continuity in the first place, duh. Does anyone know what/how to see the additional functionality you get from using the debug pin? The wiki state you get improved timing and screen positioning but I haven't seen much difference.

YoImLee, 2021-06-10:
It seems to be required for the clock generator mod to work properly, if you leave it disconnected you get a message saying "sync skipped sfr wrong: 0.0" and the clockgen wont do anything afaik.

closed 2021-11-04 by shawneeunion