ramapcsx2/gbs-control #242: Tearing after clockgen mod on 576i pal ps2

Mr12mill, 2021-06-08:
Hi.

I have installed the clockmod on my gbs-control unit.
Occationally i get a static screen tear which i understand is normal, and that goes away with a restart.
But when i use my PS2 (pal region) and play games in 50hz 576i, the line is slowly moving upwards.
With a restart the line is most often off screen.
But the problem is that because it moves it will always come back onto the screen again, wandering slowly upwards.
The movement is really slow so it can at first be mistaken as a static line (something like 20 minutes to cross the screen).
It should be noted that games, on this same unit, that support 60hz 480i only gets the static line.

I have tested with a clock generator from a different manufacturer and changed all the wires but no difference.
The clock generator is identified by the system when i activate framelock.

YoImLee, 2021-06-10:
I've pretty much given up on using the clock generator for now as it also does the slow tearing for me across 3 different GBS boards each with their own clockgen and ESP module. Tried a whole list of things to no avail and have come to the conclusion that it's most likely a firmware issue. Happens on all resolutions and both 50/60Hz, progressive/interlaced sources for me. What's confusing though is that the vast majority of people does not seem to have these issues at all, only a small group.

For now I'd omit the clockgen and just use the GBS without it as the tearing you get then scrolls by much quicker and is less of a nuisance, at least for me. Hope someone is able to figure it out though, just hard to diagnose since it seems to happen at random.

Mr12mill, 2021-06-10:
Thank you. Was about to order and build another unit, but I would probably get the same issue with the new unit.
I'm using only YPbPr input, I don't know if you have tested other input solutions, if they act the same?
I have tried both with and without board shielding with copper tape, but did no difference.

YoImLee, 2021-06-10:
Yeah, I mostly use the RGBs input but have tried YPbPr with my Wii and I get the same result.

Here are all the things I've tried already:
- PS1, PS2, SFC, N64 through RGBs
- Wii through YPbPr
- All the recommended and optional mods (extra bypass caps, replacing c11 etc.)
- Adding a 1nF cap between the clockgen's output and pin 40 on the scaler
- Reflashing the esp8266, making sure everything is correct
- All the different resolution presets
- VGA directly to a monitor
- VGA to HDMI dongle
- Several different power supplies & battery packs between 5-12V
- Different outlets throughout the house & different TVs
- Removing the heatsink from the scaler & positioning the clockgen board elsewhere
- Probably more I can't think of right now

Mr12mill, 2021-06-10:
Has anyone had any luck solving this issue?

TioChico27, 2021-06-18:
Same issue here. No luck solving this.

TimbleThimble, 2021-07-07:
Similar sorta issue here too

someitalian123, 2021-09-29:
I only use ntsc but I still get screen tearing, at any resolution, with the clock gen installed. The tearing does not happen immediately, it might take a while. I know the clock gen is installed correctly because when I select frame time lock I get the message "Active FrameTime Lock not necessary with external clock gen installed.” One workaround that I know of is to power cycle. I wonder if more shielding on the cable connecting the clock signal to pin 40 would help, or if the heat sync is too large causing too much of a gap between the clock signal and pin 40?