ramapcsx2/gbs-control #243: Request for 1080i / 540p output preset

jortego128, 2021-06-11:
Requesting a 540p or "progressive" 1080i output mode for use with early 2000's era HD CRTs, namely the Sony Hi-Scan and XBR series.  The reason is, when the HDPT option in service menu is enabled, the set enables HD Pass Through and accepts [~33.75Khz] 1080i (tested and [verified](https://i.imgur.com/WiuPzu4.jpg)), and (I'm told) 540p and displays its directly in a zero-lag, zero processing mode.  With direct fed signal, at stock svc menu settings, these sets normally have unavoidable input lag as measured by a Time Sleuth as follows:

240p/480i - 48ms at top left corner
480p - 14.7ms at top left corner
720p - 14.7ms at top left corner
1080i - 31.3ms at top left corner

With the HDPT option enabled, lag changes only for 1080i as follows:

1080i - 0.78ms at top left corner.

Currently, the 480p output of the GBS-C is good enough to obtain ~15ms of lag for all sub-HD input resolutions, but if someone were to implement the requested 1080i/540p

HyperionAlpha, 2023-01-09:
I would really like to see this feature as well.

islandniles, 2023-03-23:
This would be amazing,  any news?

ragingchaos, 2024-01-18:
Bump for this request, we're heading into an HD CRT resurgence and this would be a nice bonus for those that find 1080i sets.