ramapcsx2/gbs-control #245: Can't compile; framesync.h error

Roy12321334, 2021-06-17:
In file included from C:\Users\me\Documents\Arduino\gbs-control\gbs-control.ino:20:
C:\Users\me\Documents\Arduino\gbs-control\framesync.h: In static member function 'static bool FrameSyncManager<GBS, Attrs>::findBestHTotal(uint32_t&)':
framesync.h:148:33: error: call of overloaded 'abs(uint32_t)' is ambiguous
  148 |     if (abs(inPeriod - outPeriod) <= 4) {
      |                                 ^

gingerbeardman, 2021-06-18:
That line of code was last modified 16 months ago, so I think it's possible to compile or it would have been spotted by now.

Please check your compilation method.

godzivan, 2021-06-18:
Roy12321334 he is right, as reported on the arduino site:
https://www.arduino.cc/reference/en/language/functions/math/abs/

this is the fix :

    int32_t ttemp;
    // allow ~4 negative (inPeriod is < outPeriod) clock cycles jitter 
    ttemp=inPeriod - outPeriod;
    if (abs(ttemp) <= 4) {
    //if (abs(inPeriod - outPeriod) <= 4) {

TioChico27, 2021-06-18:
I got that same error yesterday. Try to use the Esp8266 version 2.7.4 library insted of 3.0.0, this resolved the issue here.

godzivan, 2021-06-18:
it's still a bug, my solution is correct

closed 2021-06-18 by Roy12321334

gingerbeardman, 2021-06-18:
Thanks @godzivan could you please submit a PR with the fix?