ramapcsx2/gbs-control #246: GBS Control remains in pass through mode

ereinhardt, 2021-06-21:
Hello,

i would like to use my old CRT monitor on my MacBook Pro (2016). In the following I have visualized my setup so that my problem can be followed better.

[image]

I give out a 640x480p output via switchResX. This would have to have a frequency of 31 kHz due to the low resolution (or I am guessing that the resolution should not be more than 31 kHz). So now I'm going to a YPbPr converter with the 480p HDMI signal. The 480p component video signal goes into the GBS 8200 with the latest version of GBS Control. I think this is where my problem begins:

In theory I want to downscale the 480p 31 kHz signal to a 15khz 240p. GBS Control also recognizes the component video signal. I can change all settings, except the resolution setting. It always stays with the pass throug setting. On the JVC monitor you can see the actual picture minimally; heavily glitched with strange color casts. My question is where is the mistake in my idea. Or why can I not switch to 15kHz with GBS Control? I'm really desperate :/

[image]

gingerbeardman, 2021-06-21:
Can you confirm every step?

Stop 1. Does the HDMI 480p from SwitchRes work connected to anything else?

ereinhardt, 2021-06-21:
Yes I can confirm that. I have now also managed to downscale it to 15khz. It was because I transmitted with 60fps and not with 59.x fps (at least I think so). But now I have the problem that I have glitches every 3 or 6 sec. Can you help me with that? I have included a video for illustration.

https://user-images.githubusercontent.com/35974708/122820641-1e6c8a80-d2dc-11eb-8850-365eb1aa8508.mp4

gingerbeardman, 2021-06-21:
Try searching for this new issue.

Some ideas:
- turn off Auto Gain Adjust Toggle
- adjusting HTotal (in development section of UI)
- change SDRAM clock speed option

ereinhardt, 2021-06-21:
a little update on the first problem. I found the problem. In switchResX there is a scaled down 480p with 60hz (45 kHz) and a> correct <480p with 59,952hz (31,475khz). The setting is named in both cases with 480p 60hz, which is why you can't tell the difference immediately.

[image][image]

Secondly @gingerbeardman :)
Unfortunately, that wasn't the solution to the problem. I just put the monitor on for about 15 minutes and then the glitches were almost gone. However, every time I restart the setup everything is back. It's really weird. If someone has any idea why this is so, I would be happy to hear from you. If I find the solution, I'll post it.

BambooShadow, 2021-06-22:
I don't know if this makes a difference, but I remember having trouble changing settings in GBS-Control (with the new interface), while the "Empty CUSTOM"-Slot was selected. I was able to simply fix this by creating a profile ("Slot1" in your case), select that profile and then apply changes in the interface.

Before (while having the "Empty CUSTOM"-Slot selected) some settings just wouldn't react to my inputs via webinterface. Maybe this helps.

[image]

You also wrote that by now, you were able to choose the "15KHz"-option in GBS-control - maybe that was because of the "wrong" profile being selected? -->maybe double-check!

About the Downscaling: I have tested the downscaling only with my N. Switch so far. I set the Switch to output 640x480. That should be output with a clean 60fps framerate. So I don't know if the 60Hz/59,xxxxHz thing actually matters.

I'm not trusting your switchResX-software completly here. Maybe really double-check with a monitor what's actually being output here, so I'm also recommending what ginger already wrote:
"Try a TV with input format display, or some other electronics that might show that?"

ereinhardt, 2021-06-23:
Yes, that's right with the costum slot. That actually created problems. That with the 60hz was an assumption that, as I mentioned in my update, turned out to be wrong. It was up to switchResX which occasionally scaled a 720p image to 640p, but the output still had a frequency of 45khz (as usual with 720p). GBS-C can only handle 31khz. I changed the settings in SwitchResX to >real< 640p and then there were no more problems (I added screenshots again in the update). My new problem is that I have strange glitches and interference for the first 15 minutes (as shown in the video above). These disappear almost completely after a while, but are still very annoying for the first 15 minutes. Now I would like to know if this is a problem with my setups or a general problem with old CRT TVs?

kjellxvx, 2023-12-05:
@ereinhardt did you get this setup working? How did you make your MacBook choose the 480p? I can't activate this option in SwitchResX...

ereinhardt, 2023-12-05:
What Mac you use, and what is the exact setup. 

If I connect the HDMI to YPbPr converter, I can select this resolution in SwitchResX for the converter. Here one which should work:
https://www.amazon.de/Component-Konverter-Ausgang-Fernsehern-Komponente/dp/B0BNLFVTPN/ref=sr_1_4?keywords=hdmi+to+ypbpr&qid=1701789052&sr=8-4

kjellxvx, 2023-12-07:
> What Mac you use, and what is the exact setup.
>
> If I connect the HDMI to YPbPr converter, I can select this resolution in SwitchResX for the converter. Here one which should work: [...]

Thanks for the reply! I'm using the following adapter. (it's an adapter that does not scale!)
https://www.amazon.de/dp/B0C88W3L9G?psc=1&ref=ppx_yo2ov_dt_b_product_details

If I try to add the custom non scaled resolution to switchResX it always says: "not installed", also after I reboot. 

I use MacBook Pro M1, macOS Ventura...

[image]

ereinhardt, 2023-12-12:
http://surajms.com/2015/04/fixing-switchresx-saving-issue/

ludzeller, 2023-12-27:
When I select 640x480 @ 60 in SwitchResX it seems to be running at 75kHz as @ereinhardt suggested. But I cannot select a proper 31kHz mode using this setup. @ereinhardt You are running a Intel-based MacOS setup, right? Could you give some more details, esp. regarding the version of MacOS and SwitchResX you are using?

Since I am using the same HDMI to Component converter as @kjellxvx I would like to give [the converter @ereinhardt used](https://www.amazon.de/Component-Konverter-Ausgang-Fernsehern-Komponente/dp/B0BNLFVTPN/ref=sr_1_4?keywords=hdmi+to+ypbpr&qid=1701789052&sr=8-4) a go as well, but it is not available anymore on Amazon and I can't find any other sellers in Europe right now.

As a plan B I think I will give [this hardware downscaler](https://www.amazon.de/gp/product/B07X5KL6B4/ref=ox_sc_act_title_1?smid=AMDO8QE0HLNJD&psc=1) a go and will try  to feed its 31kHz component output into GBS-C. If somebody can source other possibly suitable HDMI to Component converters then please let me know.

Is it actually the case that GBS-C can only do the 31kHz to 15kHz conversion when going into it via the component inputs? Or has somebody been able to use the VGA input for that succesfully? If so, please post your detailled setup.

My setup:
MacBook Pro 14" 2021, MacOS 13.5.2, Apple Silicon
Registered SwitchResX, 4.12.2, Apple Silicon
McBezel ODV GBS-C unknown firmware

My signal routing:
MacBook HDMI Out -> [HDMI to Component](https://www.amazon.de/dp/B0C88W3L9G?psc=1&ref=ppx_yo2ov_dt_b_product_details) -> GBS-C -> (nothing afterwards yet, but I plan to go into my Commodore 1084s via a self-made RGBHV cable)

ludzeller, 2023-12-30:
As a follow up to my previous post I can confirm that I was able to get a 240p picture out of the described setup. So apparently it is not always possible to get a suitable 480p @ 59.x signal out of MacOS and using a downscaler in between can fix this. 

Note: You will need a CRT that accepts HV sync, since GBS cannot output composite sync.