ramapcsx2/gbs-control #247: [REQUEST] OLED display tweaks and improvements

TioChico27, 2021-07-03:
I tried the OLED display and works how it should but i think it can be improved.

1: Would be nice if you can change the profile direct from "current settings" screen just turning the rotary encoder and clicking to load it;
2: Show the profile names used in web interface insted of numbers 1-7.
Is it possible to do that? Would be nice.

gingerbeardman, 2021-07-04:
@TABYDACAT

TABYDACAT, 2021-07-05:
Thnx for trying out the OLED, glad everything is working!

1. Didn't want to put anything inside the Current Settings option to avoid any confusion. More so acts like a confirmation submenu that everything you picked out before-hand actually set correctly. This _could_ be done though, but just feel like profiles don't have a place inside this submenu.
2. Technically yes, but limited space on the OLED, users would then have to think about how long they name their profiles. Basically, font alignments/configurations would probably require some major stuff to make sure names wrap correctly.

3zet, 2021-08-04:
Hi, 
I will put it here. I made much better schematics with pull up resistors and proper capacitors. 
Now should be much more stable. 
[https://i.postimg.cc/bNGM25SN/screenshot-308.png](url)

Nuck-TH, 2021-08-05:
Why you pull encoder to 5v? It is only works with ESP, so it should be pulled to 3v3 like display and with smaller value resistors to pins.

3zet, 2021-08-05:
Honestly, does not matter, gbs has both voltages. For me 5V is more convenient. For someone maybe not.

TABYDACAT, 2021-08-05:
Yea I tried to keep it simple as possible so many ppl can build without too much trouble. So the less wires, components, smaller schematic etc the better while keeping everything working. It's a constraint I put on this add-on early on hence the simplified circuit. Of course it can be improved, but this makes it so it overshoots the goal. The encoder as it stands is pretty stable in combining the ISR and pullups/caps. I just see redundancy in this schematic to where it can't accomplish that much more stability for a mechanical encoder/oled plus considering the application its being used for.

Don't get me wrong some of these things can be implemented and I agree with, but it's a hard sell. I'll revisit this at later time with your suggestions for sure though. Thanks for the suggestions!

dekkit, 2021-11-06:
@TABYDACAT 
Firstly great job with this - it really helps makes gbscontrol complete and is really easy to install.

Further to above, would adding a profile ' nickname' field (6 character max)  in web ui help in addressing this need? (Without messing with the web ui hopefully or added to the space needed in esp)

Also. few more suggestions after building it (if ok to place here)
- add an "Output" menu item - with options:
VGA
Component

- instead of defaulting to opening the menu list, auto open the 'current settings' (as this would be cool to see the system info after the gbs logo). The user can then access menu by clicking on rotary or leave it as is.

For the 'current settings' view, it would also be super handy to make it alternate each few seconds between showing the input signal info and then showning the output info.   This would help with quickly working thru issues from bad inputs etc

TABYDACAT, 2021-11-06:
Not quite understanding what you mean with the "nickname" section. Can you elaborate a bit more please? 

Output stuff is reasonable to add, good idea really, as I don't use different outputs and completely forgot to add this.

I've been working on 'Current Settings' as being the main menu, along with Scanlines/Strength and Deinterlacer options to the menu. So, hopefully soon your suggestion will come to light!

dekkit, 2021-11-07:
[image]

In the web ui for example adding a feature to give an extra "short" nickname for the save profile (in addition to the existing profile name).  This short nickname would only be used for the oled screen.

All that said, i really like your approach - the oled screen for the common / most used features and leaving the advanced setup stuff for the web ui to avoid cluttering the oled menu system.

Keep up the awesome additions, happy to test out any updates.

timotchee, 2022-08-24:
Hey @3zet, I’m likely to use your circuit design, but don’t know much about electronic circuits.  Have there been any revisions?
Also, I’d like to confirm:
1) C1 and C2 are 10 NANO faradays?  What voltage?  If I switch to 3.3V power, do the values change?
2) If using 3.3V for power, what resistors need to change?  R3, 4, 5, and 6?  R1 and R2?  Is the appropriate resistance value for 3.3V power 6.6kOhm?
Thank you!

PSHarold, 2023-01-24:
#403 This might be helpful.