ramapcsx2/gbs-control #249: Error after Si5153 Clock Generator install with GBS 8200

Veridor, 2021-07-07:
After wiring up my Si5153 clock generator board to my GBS 8200 as shown o the wiki, I confirmed continuity across all wires and when the unit is powered on, my meter shows 3.3 Volts DC across the power wires. However, when I go to test by enabling the Active Frame Lock feature in the GBS-C web UI,I get this error with the debug console enabled: "Warning: Check VDS_VS_ST!". Any clues what is going on with this error?

gingerbeardman, 2021-07-07:
Hmm. Looking at the code that error shouldn't be reached if external clock generator is used.

https://github.com/ramapcsx2/gbs-control/blob/b9265e4025730aff7e0a4dbb82ac862eb9ba1b6a/gbs-control.ino#L768-L789

Veridor, 2021-07-07:
> Hmm. Looking at the code that error shouldn't be reached if external clock generator is used.
> 
> So, I think you need to confirm your clockgen is installed and recognised correctly.
> 
> https://github.com/ramapcsx2/gbs-control/blob/b9265e4025730aff7e0a4dbb82ac862eb9ba1b6a/gbs-control.ino#L768-L789

That’s good to know. All my wires test fine for continuity and I reflowed the solder joints just in case. The board seems to be getting power too. Any wires or solder points in particular I should focus on? Is there a way to rule out a bad board without just buying another one? I tested the SMD resistors on it and it seems one is out of spec at 6k Ohm. Not sure if that would cause this issue though.

Kumulus, 2025-03-02:
Hehe got same error, googled my way here and got same mistake. Now working.

Thanks erloncabral!

Btw as usual i messed up and burned away the contact of PCLKIN of the Trueview. Was digging abit in the PCB but found  no trace so i just solderd the wire from External clock and was hoping the best. (Think i destroyed the neighbor Pin of PCLKiN aswell) .
So even then i got a picture now (with and without working external clock).