ramapcsx2/gbs-control #250: Clockgen preventing 'Force PAL 50hz to 60hz' at 1080p preset

TimbleThimble, 2021-07-12:
So cutting #155 and #237 as short as possible, I (and these other people) are having issues when trying to use PAL 50hz output forced to 60hz _specifically_  at 1920x1080. 

For me, it works fine when forcing PAL 60hz on 1280x1024 but not with 1920x1080 because (from what I can gather the clockgen is preventing it.) Display choice isn't the issue. I can display 1280x1024 50hz perfectly.

When the clockgen is involved, setting it to 1080p with force 60 toggled-on _resets_ the clockgen during the process at the _wrong_ frequency or atleast to my understanding. That is what I believe it is doing in the screenshots provided. 

[image]

Soft-resetting will then allow it to work at the correct 60hz, but with the trade-off of it no longer recognizing the clock-gen which causes other issues. 

[image]

Power-resetting will put it back to 50hz, even if the toggle is still on, even if a preset has been made, no matter what it will go back to 50hz.

[image]

I hope this doesn't sound too much like complaining but like the OP of #237 it's a bit annoying trying to find a solution other than 'buy a display that works with 50hz.' Furthermore, I'm not sure if this is something that could even be fixed in the CFW. If it is needed to elaborate further I am happy to help.

ramapcsx2, 2021-07-12:
Coders: Look for "GBS_OPTION_PALFORCED60_ENABLED" throughout the gbscontrol.ino file for all the references.
It probably doesn't have code for calculating a 60Hz timing...
Thinking about it, that will be quite hard to do, too ><

TimbleThimble, 2021-07-12:
I wish I could help more in the coding aspect but having this properly working would be a big benefit to me personally who plays a lot of PAL PS2 games.

Just to add, I hope this isn't bordering on feature request but even with forced pal 60hz 1080p (image below excluding clock-gen) the image is initially too tall to properly fit the 1080p frame without adjusting any of the scaling options. (Shown below)

[image]

I figure there is a reason for this (pardon my ignorance), more-so just curious as to why this is and what to do. I'm fine with using the scaling options but then I'm unsure/worried about maintaining proper geometry/shapes being correct, on par to the 1280x1024 preset and I'm not sure how to scale properly to keep that proper shape.