ramapcsx2/gbs-control #251: Suddenly unable to connect to GBS (In AP mode) regardless of device

TimbleThimble, 2021-07-20:
Apologies for one issue after another but I'm suddenly unable to connect to my GBS's Wifi Module no matter whatever I do. I've tried using multiple devices, power sources, changing locations to try and remove interference, reflashing the firmware (which I can do just fine, showing that the part itself still works) and it still appears on my mobile devices as a device to connect to but when I try to connect I get the generic 'unable to connect message.'

I bought a preassembled unit because I'm not too qualified for physical modification but everything looks fine regarding soldering points and it _was_ working before a few days ago. Nothing changed it just suddenly stopped working and unsure of what to do.

Any help would be appreciated.

TimbleThimble, 2021-07-20:
**To add on:** say the situation is the the NodeMCU has died in some way, I'd like to keep using the gbs, since I've reflashed and set to default I have no way to select my preferred video-mode. So is there anyway I can change the video mode defaults in the code of the firmware before flashing?

For example, the default for 480i/480p is 1280x960, is there anyway I can make the default 1920x1080?

gingerbeardman, 2021-07-20:
How long are you leaving it after power on before you try to connect?

Sometimes it can take a minute or three to settle down.

TimbleThimble, 2021-07-20:
I've left it for upwards of an hour and no luck.

Also I know last preset is default but the problem is since I reflashed it, the default went back to 960p (which doesn't fair well for my monitor and capture card) and I can't connect to change it. This is why I asked if it would somehow be possible in the code/flash process to change the default.

TimbleThimble, 2021-07-20:
Further update: Seemingly my GBS has just offed itself. Despite still powering, light on Wifi Module light identifying signal and  multiple reflashes my GBS now just doesn't display any output signal at all.

TimbleThimble, 2021-07-21:
Is there anyway I can control the GBS through USB? Like access the web menu through USB connection?

TABYDACAT, 2021-07-22:
When flashing new firmware have you also tried setting to: 
Erase Flash: "Sketch & Wifi Settings"?

TimbleThimble, 2021-07-22:
I've tried both sketch & WiFi settings and all flash contents.

leonkiriliuk, 2021-08-10:
Take a look at what your IP is.  For example, if the AP gives you IP 192.168.4.125, then in the web browser go to: http://192.168.4.1  GBS-Control GUI should come up.  The Wiki doc does not work in AP mode, it seems to now work only when GBS-Control is connected to your router.

TimbleThimble, 2021-08-11:
You misunderstand, the GBS has been re-flashed and thus isn't connected to my router and since I cannot connect to it during AP mode, like it's not even a visible device to choose from on any of my devices network settings I am un-able to even get to the point of connecting it to my router.

leonkiriliuk, 2021-08-11:
When first flashed, can you see the GBS as a router? Can you connect to it? If you can connect to it, what IP are you assigned?

leonkiriliuk, 2021-08-11:
The ESP module is a very inexpensive device.  I suspect yours, for some reason, failed.  Get another one for a couple of dollars and try again.  I think you have a hardware failure.

TimbleThimble, 2021-08-11:
While I hope this doesn't sound like a cop-out; the problem is I'm not too well versed with the de-soldering and re-soldering required to replace the module. My unit is pre-assembled. I don't have the tools nor experience doe learning would be the solution to this and the only reason I nag for some sort of USB control is that it would save me from doing so.

As for this thread, I'm just gonna close it. It's clear something within the ESP has most likely failed and I'll just have to live with what I've got I suppose.

closed 2021-08-11 by TimbleThimble