ramapcsx2/gbs-control #253: Sync gets interrupted when screen goes full white on Wii with component amp distributor

TyMiles2012, 2021-08-01:
In my particular setup, I'm having an issue where the sync quickly drops when the screen goes full white on the Wii specifically when it's plugged into my component amp (Which I use to send a component signal to multiple outputs).  I can consistently replicate this on Ocarina of Time for the Wii virtual console by using the Farore's Wind item or warping with an ocarina song.  This does not happen if the Wii is plugged directly into the scaler.  This has also not been an issue with the OSSC.  The video below shows the sync drop happening consistently with the developer's mode output enabled in a browser.

Setup:
GBS model: HCKJ 8200 v5.1 with yellow buttons, C11 replaced with 22uF capacitor.

Wii > Component amp using HD Retrovision cables > One of the outputs into the scaler via component.

Every possible form of output has this issue, including YPbPr mode.

The specific component amp is the Stellar-Labs 33-12580

[https://www.youtube.com/watch?v=BHW57eRelnM](url)

YoImLee, 2021-08-01:
This happens to me on PS1 also. I fixed it by replacing C33 and C35 with some 1nF 0603 ceramic capacitors as per this guide on the wiki: https://github.com/ramapcsx2/gbs-control/wiki/Sync-on-Green-Capacitor-Replacements

TyMiles2012, 2021-08-02:
That fixed it!! Thank you!

closed 2021-08-02 by TyMiles2012