ramapcsx2/gbs-control #254: Can't access web ui

Retron101, 2021-08-03:
I just finished building the gbs control but I have a hard time connecting to the web ui. Almost every time I try to go the the web link while connected directly to the esp8266 it says that either the dns is not working, ip doesn't exist, and/or no internet. What can I do to fix this issue?

Sirethief2525, 2021-08-14:
I bought mine pre built and now I can't access the ui, whether it be through its on board WiFi or my own router it will not connect. I need a master reset or something but can't access the menu to change anything. Any help would be appreciated

Sirethief2525, 2021-08-14:
Never mind, for whatever reason after I posted this it decided to start working again.