ramapcsx2/gbs-control #256: Arduino Error: SSD1306Wire.h

segacollection, 2021-08-05:
I'm trying to Verify/Compile and always have this same error:

```
gbs-control:23:10: fatal error: SSD1306Wire.h: No such file or directory
   23 | #include "SSD1306Wire.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
SSD1306Wire.h: No such file or directory
```
any experienced users able to help?

tomudo, 2021-08-06:
It mean missing SSD1306Wire library install. SSD1306Wire library use for controlling LED display which this feature added later. You may install SSD1306Wire library or use GBS-Control older version.

segacollection, 2021-08-06:
Thank you. How do i install that library?

tomudo, 2021-08-06:
The port_to_input_PGM function not found in your compiler environment. I added SSD1306Wire lib and compiled the latest without problem. Maybe your Arduino ide setup incorrect in some way.

segacollection, 2021-08-06:
I am relatively new to arduino; I set up by downloading the installer and following instructions, really. Any tips on where to look?!

segacollection, 2021-08-06:
I bought [this board;](https://www.amazon.de/gp/product/B07CVW85W4/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) should i still check the Lolin D1 R2 & mini?

segacollection, 2021-08-06:
When I plug the ESP8266 into USB the LED does not remain on. Should it?

closed 2021-08-06 by segacollection