ramapcsx2/gbs-control #257: Clock Generator Mod screwing with colours

tunnotron3000, 2021-08-08:
Hi there, to start with, I'm sorry but I don't have a log to share as I was getting in such a fluster with this whole thing.

I set up my GBS Control yesterday, everything all good, no issues whatsoever. Recorded around an hour of gameplay and altogether just played the hell out of some games. Then today, I decided to record some more gameplay from my Megadrive and this started happening:

https://www.youtube.com/watch?v=Rpt2UetacqE

The colours get all messed up after a short amount of time. This is via RGB. The same thing happens albeit with purple instead of green for component.

I've reflashed the ESP8266, resoldered everything and yet the issue persists. The only thing that fixes it is uninstalling the clock gen mod. It seems to happen whenever these messages appeared:
`h: 433 v: 523 PLL:5 A:4a4a4a S:8f.10.00 H-   I:00 D:b637 m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-74
h: 433 v: 523 PLL:5 A:4c4c4c S:8f.10.00 H-   I:00 D:80ff m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-73
h: 433 v: 523 PLL:5 A:4e4e4e S:8f.10.00 H-   I:00 D:35b6 m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-73
h: 433 v: 523 PLL:5 A:505050 S:8f.10.00 H-   I:00 D:0000 m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-73
h: 433 v: 523 PLL:5 A:525252 S:8f.10.00 H-   I:00 D:7e00 m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-73
h: 433 v: 523 PLL:6 A:545454 S:8f.10.00 H-   I:00 D:80ff m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-73
h: 433 v: 523 PLL:6 A:565656 S:8f.10.00 H-   I:00 D:be00 m:1 ht:2558 vt: 259 hpw: 235 u:  0 s:ff S:13 W:-73`

I'm just wondering whether I have a dodgy clock board or this is a common issue?

tunnotron3000, 2021-08-08:
Just as an addition, I reinstalled the mod as I was concerned maybe the heatsink was shorting something. Totally reinstalled and the same thing happened. I've ordered another board and will update.

TABYDACAT, 2021-08-08:
No one has brought up this issue a lot at least in connection with the clock board mod. Does the confirmation message come up with the clock board mod installed? Is there something in between the clock board and heatsink to act as insulation? By your message it sounds like your installing the clock board right on top of the heatsink which is the only reason I ask.

tunnotron3000, 2021-08-09:
Yeah everything seemed to be fine. I reinstalled the board again with electrical tape on the bottom as well as extra foam to eliminate any chance of shorting and the issue remains. I triple checked the wiring too. I'm hoping that it's just an error with this particular clock generator unit. I mean, if I remove the unit and the GBS works fine (with tearing) then the issue must be with the clock gen board, right?

tunnotron3000, 2021-08-11:
Okay, well I think this can be closed off now. After much, much googling around I realised I'd wired the 5V wire to the 3v3 input. I had to reorient the ESP board anyway so completely redid my wiring. I've been letting the GBS run for about 10 mins and so far the colours haven't changed so I'm thinking that must have been the issue - maybe when the power is too low for the ESP board it fudges up the colours?

closed 2021-08-11 by tunnotron3000