ramapcsx2/gbs-control #261: RGB Full/Limited Range Option

ShougSCS, 2021-08-21:
I don't know if it could be implemented, but I would like for there to be a toggle between RGB Full/Limited range for different screens. I use my GBSC on my computer monitor, which is full range, while the GBSC itself seems to be limited range, causing a washed out picture. My monitor doesn't have the ability to change RGB ranges, and I would like for there to be some way to change it within the GBSC.

ramapcsx2, 2021-08-21:
Hm, the analog VGA output is always what's considered full range.

ShougSCS, 2021-08-24:
It might be the converter I'm using. I can confirm that it's outputting in limited range, as when I set OBS (a program that I typically have set for full range and had a similarly washed out picture) to limited, it suddenly looks perfectly fine.

someitalian123, 2021-08-27:
On my tv, I think I'm getting color crush on areas that are too bright or too dark. I don't believe my tv actually supports full, either that or the vga-to-hdmi adapter I'm using doesn't. Either way, if there were a toggle I think it would help.

johngarcia000, 2023-12-08:
Did you solve it? I got crushed black too.