ramapcsx2/gbs-control #263: Lost C14 capacitor

grimkor, 2021-08-30:
Firstly: what a project!

So I've gone and lost the solder pad for C41 and C41 itself, blame my incompetence with tiny SMD components.

The setup is still working but I'd like to get an idea to where C41 connects so I can look to remedy it if needed.

My continuity test shows that the outer side of C41 connects to C61, is that right?

jcpurdy00, 2022-03-21:
This happened to me too. Did you ever find an answer to this?

alextwothousand, 2022-12-02:
I'm trying to find someplace that I can replace C41 with. Its not easy to solder to.
[image]

ramapcsx2, 2022-12-03:
It's a power supply bypass cap for one of the 2 power planes. It's certainly not critical, as there are the many others, and especially if you've beefed those others up :)