ramapcsx2/gbs-control #265: Allow to control this device completely via OSD and allow to use Arduino Nano again

M-Reimer, 2021-09-02:
I came to this project from some YouTube videos where the scaler, based on the GBS mainboard and the GBS-Control firmare was tested.

I need a simple scaler to use a SNES, N64 and probably a NES on a "modern" LCD TV.

BUT: For me it just feels weird that I have to use an external device (Smartphone, PC, ...) to configure it. I would really prefer being able to do the configuration directly on the GBS mainboard. After all it's just a scaler, I don't plan to configure this regularly. Best case I'll do this once and forget about it.

I've seen that there is some OSD code which seems like a "proof of concept" attempt without covering most settings.

- Would it be possible to extend this to cover everything the web interface has to offer?
- Which documentation would be needed to do that? Where did developers here get all the ugly HEX codes from that you send to the scaler chip?
- Just assuming that I would find the time and would be able to extend the OSD, are there any additional reasons why the firmware, in its current state, can no longer run on the Atmega328p (Arduino Nano)? I would really prefer to use a controller without Wifi support.

tomudo, 2021-09-05:
Arduino base microcontroller has too little flash memory to achieve a certain code scale. You should try to modify the very old 2 years arduino gbs-control to see the reason to use ESP base microcontroller.

M-Reimer, 2021-09-05:
I had a deeper look into the code in the meantime. Seems like at some point many features exclusive to the ESP controllers have been implemented which can't be just mapped to the atmega controllers. One example would be SPIFFS which does not work on the limited EEPROM of the atmega chips.

I'll at first build one device as intended here and give it some time to see how I like it.

If I, at some time, decide that I want to try the OSD approach I would most probably just fork this project and drop everything regarding the web interface and OLED support to reduce the overall code base a bit. For a start I would also completely drop profile support. For an atmega chip this has to be reimplemented based on efficient structs to save the data in binary form with fixed spacing to EEPROM.

closed 2021-09-05 by M-Reimer

ramapcsx2, 2021-09-05:
A classic Arduino (ATMega) chip can't do many of the advanced features anymore.
The concept of adaptive presets, so the entire preset system really, doesn't make sense without it.
This project should really use the much more advanced ESP chip. It's the same price, but better.