ramapcsx2/gbs-control #266: Why do you place the frequency generator on top of the heatsink

M-Reimer, 2021-09-10:
I've been searching the web extensively and all I could find that at some point it was done for the first time and from that time on everyone places it onto the heatsink.

Is there any reason why this is the preferred location? After all it's a heatsink and having a board in its way will not be too good to make it do its job.

I'm wondering if I could place a smaller board directly to the location where the potentiometers used to be? Any downsides of doing this?

Sorry for posting this question as Issue. Could you enable the GitHub discussions feature for this project?

ramapcsx2, 2021-09-10:
It's a convenient location for a few reasons. Mainly it shields any clock noise from the analog parts in the ASIC, and secondly, it provides a good anchor point to screw in the board.

The heat sinking isn't required much. I've ran the scaler without any heatsink at all, and while not recommended, it worked just fine for hours. (It did get quite hot, but then stayed at that level.)

M-Reimer, 2021-09-10:
Will give it a try if it has any disadvantage to mount a board where the potentiometers were mounted.

But now I have to get this ding to do something first. Already wasted half a day and can't get my SNES to output anything throuth this board. Most settings, for whatever reason, don't toggle at all in the web interface and all I can get this thing to display is a solid black screen :cry:  Currently I have no clue what is going wrong and will give up for today. Maybe I have more success tomorrow....

M-Reimer, 2021-09-10:
Created a new issue for my "black screen" problem: #267

wasbaun, 2022-03-21:
> Will give it a try if it has any disadvantage to mount a board where the potentiometers were mounted.
> 
> But now I have to get this thing to do something first. Already wasted half a day and can't get my SNES to output anything throuth this board. Most settings, for whatever reason, don't toggle at all in the web interface and all I can get this thing to display is a solid black screen 😢 Currently I have no clue what is going wrong and will give up for today. Maybe I have more success tomorrow....

Hi, did you notice any downside plcing the clockgen on the potentiometer location? Im planing to use this beceause, putting something on a heatsink lets me sleep not well ;-)

M-Reimer, 2022-03-23:
[image]
You can seem how I mounted the generator in this image. To create the needed distance, I've added some FR4 (PCB board material without any copper on it) between the GBS board and the frequency converter board. I also had to cut a little bit on the lower portion in the image to make the generator board move a bit closer to the pin I actually want to solder to. If you do it that way you can just stick one wire for ground through to the GBS board which provides good low impedance ground coupling.

M-Reimer, 2022-03-23:
Wiki updated.

https://github.com/ramapcsx2/gbs-control/wiki/Si5351-Clock-Generator-install-notes

So this Issue can be closed now.

closed 2022-03-23 by M-Reimer