ramapcsx2/gbs-control #267: Black screen only

M-Reimer, 2021-09-10:
After half a day of fiddling I still have no clue why I get absolutely nothing from my SNES.

I've hooked up RGB, a GND line (verfied that it is actually connected in the Scart plug) and the "Sync on Composite" from the SNES to the "S" line (hope that's composite sync).

Then I connected everything as described in the Wiki.

If I power everything, then I have one LED on the GBS-8200 which lights in red all the time and one on the NodeMCU which is blue all the time.

Only very few of the "toggle switches" in the Web GUI work. Most of them just stay off. And all I can get on my TV is a completely black image. Is there any way to find out what is going wrong?

M-Reimer, 2021-09-12:
OK, the image is not actually "black". If I turn the brightness of my TV up quite a bit, then I see a faint picture with the correct resolution.

If I disable gbs-control completely, then all I see with my current input configuration is a message "No signal".

ramapcsx2, 2021-09-12:
I'm sorry, I can't tell what the issue might be. I suspect a problem with the TV, maybe because of the PAL format..
The logs look good. You should get a picture.

M-Reimer, 2021-09-12:
Can't be the TV. I use a VGA to HDMI adapter and verified that it works on the HDMI port of my Laptop. I start to think that the input stage of my GBS board may be broken. Should my SNES work on the GBS without your firmware.

What about the broken settings switches in Web UI. Is this known?

M-Reimer, 2021-09-13:
If my board is dead and I decide to order a new one, then I need some way to test the new one in advance before doing any mods to it.

A few minutes ago, I just took a regular VGA cable and hooked my laptop to the input of the GBS board. It actually got detected as a "unknow screen" and I was able to choose 800x600 as the smallest possible resolution with 50 Hz refresh rate. If I do so, then the "stock firmware" on the GBS board provides me with a green image (whole TV screen covered in a green color with some interference patterns). This image regularly cuts out. Is this expected behavior?

If you don't have broken features in the web UI, could you please compile a version on your system and attach the binary here, so I can upload it via esptool? Just to rule out any library version or compiler issues.

ramapcsx2, 2021-09-13:
I'm really sorry but I don't have time for gbscontrol right now. I can reply to a few questions here and there, but that's about it.

M-Reimer, 2021-09-15:
No problem. Thanks anyway for your answers.

I actually get another try now. I'm pretty confident that the board was dead on arrival and the seller was OK with the fact that I've added two pin headers. Probably my returned board will be binned anyway. New board is on its way.

M-Reimer, 2021-09-16:
OK. As the new board had the same problem I had a deeper look into my connections and found the problem. I was one pin off on the Scart socket so the RGB was only coupled in capacitively.
I have sync problems but I think this can be fixed using an LM1881. I've ordered one and will create the circuit on a breadboard, first, to see if this helps to fix this issue.
I'll keep both boards, now. If I manage to get a "nice" result, I can maybe give away one finished scaler.

closed 2021-09-16 by M-Reimer

ramapcsx2, 2021-09-16:
SNES has tricky sync to begin with. It might need a bit more attenuation on the Composite Video (for sync) signal.
An LM1881 will do the trick though.