ramapcsx2/gbs-control #268: A few questions

dabilde, 2021-09-12:
I'm about to buy the parts for a GBS-C build. I didn't know where to ask these questions so sorry if this is not the best place to do it.

I'm interested in using only YPbPr component for inputs (PS2 and Wii), so I don't need to install a  RGB Scart input port. I want to do the most basic install possible. Looking at the hardware mod library, I see at least one mod that shouldn't be required for a build with only component inputs, the 100ohm across sync and ground / R34 75ohm, but it is listed as an essential one.

Basically, my question is: for the most basic decent build (component input, VGA output), would removing the rgb pots and installing the WiFi module + the clock generator be enough? Or am I missing something that is required/essential for functionality/good picture quality?

And also, is the GBS-C capable of bypassing the component input of a PS3, for example? (at 1080p) Obviously its image doesn't need to be upscaled but it would be practical for me cause it means one less hdmi input taken on my TV.

Thanks in advance.

ramapcsx2, 2021-09-12:
If you never want to use the RGBS input, you can skip the 100 Ohm resistor (since you won't need that sync source), as well as the potentiometer removal (since you won't need the RGB source).
I honestly forgot what the bypass performance was like for different sources. You'll have to test whether it works :p

dabilde, 2021-09-12:
Thanks for the quick reply. 

Awesome, so I just have to install the ESP8266 module and the Si5351 clock generator? Meaning, solder 5 wires for each?
And if I get a yellow button GBS-8200, remove that one SMD capacitor?

closed 2021-09-15 by dabilde