ramapcsx2/gbs-control #269: Will my GBS board work with this mod? It is "V5.1"...

Gabriel2Silva, 2021-09-12:
[image]

So, recently I got this GBS board from AliExpress. It works unmodded fine with a Nintendo Wii. I was reading up on the mod and I noticed my board isn't exactly the "yellow button" board, it isn't V5.0, but V5.1, and it has "20210115" written on it, so it's probably a recent model. It looks undocumented. I just want to know if I still need to remove C11. Thanks in advance.

TABYDACAT, 2021-09-14:
Follow "5.0" part of wiki of removal/replacement

ramapcsx2, 2021-09-14:
The board will work, but the quality will possibly not be any good. Trial and error (but get the recommended board soon).

TotalCoolness, 2021-09-14:
> 
> 
> The board will work, but the quality will possibly not be any good. Trial and error (but get the recommended board soon).

Is 4.0 the recommended one and is 5.0 / 5.1 inferior ?

askotx, 2021-09-15:
I get a board like that, even same date, it's working and I ask the same question.
[image]

Gabriel2Silva, 2021-09-15:
Modded it and it is working. I have removed C11. Only issue I have is that the gbscontrol keeps going back to Access Point mode randomly, but I'm not sure if that's related to the board version.

ramapcsx2, 2021-09-15:
The output section of the yellow button boards is missing filters, so there's a greater chance of a noisy image.
If yours is fine, then great :)

askotx, 2021-09-16:
Confimed, removing C11 make picture looks clearer, a bit but better.